prusa3d/Prusa-Firmware #4670: VSCode can't find python.

RogerInHawaii, 2024-04-19:
VSCode can't find python.

I'm trying to run VSCode on my Windows 11 computer.

I've followerd all the instructions for installing VSCode on Windows but when I start it up and select the folder for my downloaded Open Source Prusa Firmware it gives me an error that it can't find Python:

[cmake] Not searching for unused variables given on the command line.
[cmake] -- Could NOT find Python3 (missing: Python3_EXECUTABLE Interpreter) 
[cmake] -- Configuring incomplete, errors occurred!
[cmake] CMake Error at cmake/Utilities.cmake:6 (message):
[cmake]   Python3 not found.

Python is indeed installed on my computer. I can do this in the VSCode terminal:

S C:\Prusa-Firmware For Gemini Version\Prusa-Firmware> python --version
Python 3.12.3

And even if it's looking for Python3, I've got it set up so that Python3 links through to the location of python, and I can do this in  the VSCode Terminal:

PS C:\Prusa-Firmware For Gemini Version\Prusa-Firmware> Python3 --version
Python 3.12.3

so it's definitely finding it python, it's just not finding it when it's going through the CMake for loading in the Prusa Firmware.

What do a I do?

gudnimg, 2024-04-21:
Have you tried deleting the CMakeCache folder?

Also is Python in PATH? The fact that `python —version` works seems to confirm it. 

Please don’t create an alias `Python3` for `python`. It is only on non-Windows system where you would use `python3` instead of `python`. 

Python3_EXECUTABLE is an internal variable name in CMake, See: https://cmake.org/cmake/help/latest/module/FindPython3.html

On my end, if Python is in PATH I have never had any problems.

gudnimg, 2024-04-21:
I tried a fresh clone, and it works fine for me 
[image]

Here is what PATH looks like on my end 
[image]

Python 3.12.3 also works fine for me:
[image]

3d-gussner, 2024-05-02:
Closing due to lack of interaction.

closed 2024-05-02 by 3d-gussner

THiNK3dAUS, 2024-12-08:
Can confirm that I was having this same issue. 
Installing Python in the PATH setup as suggested by @gudnimg has fixed it, and cleared any 'not found' errors.