prusa3d/Prusa-Firmware #4672: Make Prusa-Firmware MK3 3.14+ compatible with older printers

pdxtigre, 2024-04-20:
Hello,

I'd like to suggest that we should remove the check for older printer types here moving forward and use only 

`current_position[Z_AXIS] = Z_MAX_POS + 2.0;`

https://github.com/prusa3d/Prusa-Firmware/blob/65fde953711a42398e646108bb34ebb9810c728c/Firmware/Marlin_main.cpp#L2241-L2246

Same here:

`current_position[Z_AXIS] = Z_MAX_POS-3.f;`

https://github.com/prusa3d/Prusa-Firmware/blob/65fde953711a42398e646108bb34ebb9810c728c/Firmware/ultralcd.cpp#L2856-L2864

The reason is that it will make this firmware compatible with even older Prusa MK2.  Let me know if that would cause any incompatibility issues.

So far, I've compiled and run this codebase on Prusa MK2 upgraded to MK25S with out any problem with those changes.

3d-gussner, 2024-04-22:
@pdxtigre Thanks for the report. Please see https://github.com/prusa3d/Prusa-Firmware/issues/4604
The hard coded values are a bit of pain. But these values above are valid as the extruder bodies and other parts of MK2, MK2.5, MK2.5S, MK3 and MK3S differ a bit.
So you can't just set "one" value for these different printers, as mentioned above the hard coded values aren't helping.

3d-gussner, 2024-04-30:
Closing as it is a duplicate

closed 2024-04-30 by 3d-gussner