prusa3d/Prusa-Firmware #4677: [QUESTION] VSCode Project version git - VERSION "...64" format invalid.

dARoB2108, 2024-04-30:
When starting VSCode or opening of the Prusa-Firmware folder, I am greeted with an error message regarding the Firmware version from git. While everything was working properly just hours ago, now nothing does. Here is the log from the output window. Can anyone help with this one?

Thanks!

Cheers,
RoB

[proc] Executing command: ${workspaceFolder}/.dependencies/cmake-3.22.5/bin/cmake --version
[proc] The command: ${workspaceFolder}/.dependencies/cmake-3.22.5/bin/cmake --version failed with error: Error: spawn ${workspaceFolder}/.dependencies/cmake-3.22.5/bin/cmake ENOENT
[proc] Executing command: ${workspaceFolder}/.dependencies/cmake-3.22.5/bin/cmake -E capabilities
[proc] The command: ${workspaceFolder}/.dependencies/cmake-3.22.5/bin/cmake -E capabilities failed with error: Error: spawn ${workspaceFolder}/.dependencies/cmake-3.22.5/bin/cmake ENOENT
[variant] Loaded new set of variants
[kit] Successfully loaded 0 kits from C:\Users\RoB\AppData\Local\CMakeTools\cmake-tools-kits.json
[kit] Successfully loaded 1 kits from e:\Projects\Prusa-Firmware\.vscode\cmake-kits.json
[proc] Executing command: e:\Projects\Prusa-Firmware/.dependencies/cmake-3.22.5/bin/cmake --version
[proc] Executing command: e:\Projects\Prusa-Firmware/.dependencies/cmake-3.22.5/bin/cmake -E capabilities
[proc] Executing command: e:\Projects\Prusa-Firmware/.dependencies/cmake-3.22.5/bin/cmake -SE:/Projects/Prusa-Firmware -Be:/Projects/Prusa-Firmware/build -G Ninja
[proc] The command: e:\Projects\Prusa-Firmware/.dependencies/cmake-3.22.5/bin/cmake -SE:/Projects/Prusa-Firmware -Be:/Projects/Prusa-Firmware/build -G Ninja exited with code: 1
[main] Configuring project: Prusa-Firmware 
[proc] Executing command: e:\Projects\Prusa-Firmware/.dependencies/cmake-3.22.5/bin/cmake --no-warn-unused-cli -DCMAKE_BUILD_TYPE:STRING=Release -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_TOOLCHAIN_FILE:FILEPATH=e:\Projects\Prusa-Firmware/cmake/AvrGcc.cmake -DCMAKE_MAKE_PROGRAM:STRING=e:/Projects/Prusa-Firmware/.dependencies/ninja-1.10.2/ninja -SE:/Projects/Prusa-Firmware -Be:/Projects/Prusa-Firmware/build -G Ninja
[cmake] Not searching for unused variables given on the command line.
[cmake] -- Project version (Configuration.h): 3.14.0.17
[cmake] -- Filename suffix..................: FW_3.14.0-BETA1
[cmake] -- Filename debug suffix ...........: FW_3.14.0-BETA1+7945
[cmake] -- Configuration.h and tag (not enforced): (3.14.0.17/...64)
[cmake] -- Commit Nr: Configuration.h: 7945 Tag: 
[cmake] -- These tag values will override Configuration.h
**[cmake] -- Project version git..............: ...64**
[cmake] -- Project version git hash.........: GIT-NOTFO
[cmake] -- Project version git description..: GIT-NOTFOUND
[cmake] -- Filename suffix .................: FW_3.14.0-BETA1
[cmake] -- Filename debug suffix ...........: FW_3.14.0-BETA1+7945
**[cmake] CMake Error at CMakeLists.txt:148 (project):
[cmake]   VERSION "...64" format invalid.**
[cmake] 
[cmake] 
[cmake] -- Configuring incomplete, errors occurred!
[proc] The command: e:\Projects\Prusa-Firmware/.dependencies/cmake-3.22.5/bin/cmake --no-warn-unused-cli -DCMAKE_BUILD_TYPE:STRING=Release -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_TOOLCHAIN_FILE:FILEPATH=e:\Projects\Prusa-Firmware/cmake/AvrGcc.cmake -DCMAKE_MAKE_PROGRAM:STRING=e:/Projects/Prusa-Firmware/.dependencies/ninja-1.10.2/ninja -SE:/Projects/Prusa-Firmware -Be:/Projects/Prusa-Firmware/build -G Ninja exited with code: 1

3d-gussner, 2024-05-02:
@dARoB2108 Just tested your issue on Linux and Windows, without any issues on both while following the instrauction in the README.md https://github.com/prusa3d/Prusa-Firmware#visual-studio-code-vscode instructions.

Also tested a download as zip and also here no issues.

dARoB2108, 2024-05-02:
I tried those steps again and created a new project after the issue appeared for the first time with no luck.
As far as I understood, CMake tries to get the version number from git and it seems that this number is not proper. How to avoid/repair it I do not know!

dARoB2108, 2024-05-02:
I'm using windows 10.
I just tried the method of downloading the zip file, unzipped it, started VSCode, open folder from new destination, started the bootstrap python code, reloaded VSCode, same issue.

But I think I might know what I have done wrong. In CMake there is this small button called "Delete Cache and Reconfigure" - what I have mistakenly activated this. How to revert back to old settings?

Thanks!

dARoB2108, 2024-05-02:
I just tried everything again from beginning with the instructions and this time everything worked flawlessly. I don't understand why, but this can be closed.

Thanks!

closed 2024-05-04 by gudnimg

3d-gussner, 2024-05-06:
@dARoB2108 Glad to hear that is solved :hugs: