prusa3d/Prusa-Firmware #4678: FW Crash Error on Startup with FW 3.13.3 (MK3, MK3S, MK3S+)

martinglass, 2024-05-01:
I updated the firmware to 3.13.3 and I'm now getting FW Crash errors on startup. It was working fine and error-free before this FW upgrade.

I'm going to revert back to 3.13.2 until I see this issue fixed.

PrinterLibrarian, 2024-05-07:
I also just had this FW Crash error. Firmware 3.13.3

martinglass, 2024-05-20:
It's been three weeks since I reported this Crash error and I've not had a response from Prusa yet. Others have noticed the same problem.
Can someone from Prusa provide some support here, please?

martinglass, 2024-05-20:
> If you want an official response from someone at Prusa then you should reach out to their technical support.
> 
> I would be glad to help you but unfortunately the description you give for this crash doesnt tell me anything. I can’t reproduce it on my MK3S+ 🤷‍♂️. It would be nice to see the logs from the printer (from the serial port). That would give contributors like myself a starting point.
> 
> You mention in the title MK3, MK3S, MK3S+. Have you seen this crash on these 3 printers? If not please be more specific. It would also be useful to know for example if your MK3S+ is stock or not (upgraded).

many thanks for your reply.

cozmo14047, 2024-05-28:
@gudnimg replying for someone on forum, fully stock mk3s+ + mmu3, after starting a print about half way through the printer reboots reporting crash in firmware detected, after down grading to 3.13.2 it works fine. When reupgrading to 3.13.3 it keeps firmware crashing.
Have requested logs to be uploaded here or to forums will let you know the response.
thanks

closed 2024-08-06 by github-actions[bot]