prusa3d/Prusa-Firmware #4680: MK4 MMU3 0,25 mm nozzle profile for PS

Tadyzz, 2024-05-02:
**Printer type** - MK4
**MMU Upgrade** - MMU3

Hello guys,

It´s possible that Prusa create 0,25 mm nozzle profile for MK4 MMU3 (for PrusaSlicer)?. Right now we are waiting for our MMU3 but after we receive that upgrade we need to print 300+ 4colour key chains. Due to the small dimensions we chose to print with 0,25mm nozzle.

3d-gussner, 2024-05-13:
@Tadyzz This is the MK2.5/S and MK3/S/+ repository. I would guess best place to request any PrusaSlicer settings is [here](https://github.com/prusa3d/PrusaSlicer-settings) and maybe [here](https://github.com/prusa3d/PrusaSlicer)

3d-gussner, 2024-05-13:
@Tadyzz You are aware that the smaller nozzle and also smaller layer height will increase the print time and wipe tower by a huge factor?
Here a short example with 5x5x5 cubes:
MK3S+MMU3
0.4 nozzle and 0.2mm layer height
[image]
vs 0.25 nozzle and 0.1mm layer height 
[image]

Tadyzz, 2024-05-13:
> @Tadyzz This is the MK2.5/S and MK3/S/+ repository. I would guess best place to request any PrusaSlicer settings is [here](https://github.com/prusa3d/PrusaSlicer-settings) and maybe [here](https://github.com/prusa3d/PrusaSlicer)
Ah ok...i had a link from prusa support..:)

Tadyzz, 2024-05-13:
> @Tadyzz You are aware that the smaller nozzle and also smaller layer height will increase the print time and wipe tower by a huge factor? Here a short example with 5x5x5 cubes: MK3S+MMU3 0.4 nozzle [...]

I know that..but i need to print 300+ pcs 4 colour key chains..there are few small details...maybe i try to make a new 3D model for 0,4mm nozzle...

3d-gussner, 2024-05-13:
> I know that..but i need to print 300+ pcs 4 colour key chains..there are few small details...maybe i try to make a new 3D model for 0,4mm nozzle...

I have found one tester having great results with the MK4 0.25 nozzle and MMU3 but that wasn't an official test.

3d-gussner, 2024-05-16:
Closing as wrong repository.

closed 2024-05-16 by 3d-gussner