prusa3d/Prusa-Firmware #4683: [QUESTION]<Enter comprehensive title>

RogerInHawaii, 2024-05-09:
I've downloaded the Prusa Printer Firmware and have been making some modifications to it and loading it into my Prusa i3 MK3 printer.

But each time I turn the printer on a message is displayed on the printer that says:

MK3 Firmware
detected on MK3S
printer

I'm guessing I did something wrong, but I don't know what.
Did I download the wrong version of the firmware somehow?

I'm hoping someone can offer some advice.

3d-gussner, 2024-05-10:
@RogerInHawaii If you have a MK3, with the Laser PAT9125 filament sensor (4 pins), you need to download the MK3 firmware version. The MK3S/+ firmware version is for the IR filament sensor (3 pins).
The firmware detects which filament sensor is installed and if you flash the wrong version it will report it on the screen.
So please verify which sensor you have installed.

Which modifications did you do to the firmware and did you use the MK3 variant file to compile it?
Which firmware base do you use to modify your custom firmware? Please use stable or rc tagged source code as the MK3 branch is the development branch and isn't fully tested until a release.

Or did you do some modifications to the hardware?

3d-gussner, 2024-05-23:
Closing due to no interaction

closed 2024-05-23 by 3d-gussner