prusa3d/Prusa-Firmware #4684: [FEATURE REQUEST]<Feature Request for MMU3 on MK4 3D Printer: Implementing a 'CUT' Function upon Filament Loading Error >

RomanSK1965, 2024-05-10:
**Objective:**
Enhance the usability and efficiency of the MMU3 with MK4 3D printer by adding a 'CUT' function that activates when there is a filament loading error.

**Problem:**
Currently, when the MMU3 encounters a filament loading error during a print job, the entire printing process is halted, and user intervention is required to resolve the issue. This disruption can lead to significant downtime and potential waste of materials.

**Proposed Solution:**
Integrate a 'CUT' function that can be manually triggered when a filament loading error is detected. This function would allow the printer to cut the problematic filament section and attempt to reload, thereby continuing the print job without requiring a complete stop and manual intervention.

**Benefits:**

Reduced Downtime: Allows continuous printing without significant pauses, enhancing overall efficiency.
Less User Intervention: Minimizes the need for users to physically interact with the printer, which is particularly beneficial in long print jobs or unsupervised operations.
Material Conservation: Prevents the waste of filament by not requiring a full reprint in the case of minor filament issues.
Implementation Considerations:

Ensure that the 'CUT' function is easily accessible through the printer’s user interface.
Incorporate safety measures to handle the cut filament pieces properly.
Provide detailed user documentation on how to utilize the new function effectively.

**Requested Action:**
We ask the development team to evaluate the feasibility of this feature and consider its implementation in an upcoming firmware update for the MMU3 and MK4 printers.

3d-gussner, 2024-05-10:
@RomanSK1965 This is the repository for the Mk2.5/S and MK3/S/+ 8-bit firmware, please reopen your issue in the 32-bit repository https://github.com/prusa3d/Prusa-Firmware-Buddy/issues and close it here.

3d-gussner, 2024-05-16:
Closing as wrong repository

closed 2024-05-16 by 3d-gussner