prusa3d/Prusa-Firmware #4685: [FEATURE REQUEST] Make fans spin in other direction when printing overhangs

svcuriouscat, 2024-05-11:
**Printer type** - MK3S
**MMU Upgrade** - MMU2S

**Is your feature request related to a problem? Please describe.**
The fans that cool the print and the extruder push down on newly extrude filament, which worsens overhangs.
  
**Describe the solution you'd like**
I want a setting that makes the fans suck the air from the bottom of the extruder rather than pushing the air down onto the print while printing spots that may produce sagging due to gravity and airflow (bridges).

3d-gussner, 2024-05-13:
@svcuriouscat 
>I want a setting that makes the fans suck the air from the bottom of the extruder rather than pushing the air down onto the print while printing spots that may produce sagging due to gravity and airflow (bridges).

This is not possible with the electronics as we can't switch the fan wires. See https://github.com/ultimachine/Einsy-Rambo/blob/1.2e/board/Project%20Outputs/Schematic%20Prints_Einsy%20Rambo_1.2e.PDF
[image]

3d-gussner, 2024-05-13:
@svcuriouscat Please close this issue.

svcuriouscat, 2024-05-13:
I see.  And what about slowing down or stopping the fans?  I understand that airflow helps with making the filament solidify, but it looks like too much of it pushes it down making it sag.

3d-gussner, 2024-05-13:
@svcuriouscat Please check PrusaSlicer -> Filament Settings -> Cooling -> Dynamic fans speeds maybe that can help you.

closed 2024-05-15 by svcuriouscat