prusa3d/Prusa-Firmware #4697: [ENHANCEMENT]<MK4 Z-height Display>

James-G4CLF, 2024-05-31:
My MK2 and my MK3 display their current Z-height while printing - this is convenient, especially when expecting a pause for nut/magnet insertion or filament change. My MK4 does not. I know I can see it on the dashboard, but often my computer is off while I am printing.

Please include this display with the next firmware upgrade.

It is possible that this feature is available - but if so I have been unable to turn it on - therefore it is badly indexed in that searches do not reveal how to enable it.

In this case please make it easier to find.

mix579, 2024-05-31:
You can customize what you see on the display footer in the LCD menu settings, including Z height

James-G4CLF, 2024-06-01:
Thank you - problem fixed, and first nut embedded on time!

But the manual should make it clearer that you can do this - and how.

closed 2024-06-01 by James-G4CLF

James-G4CLF, 2024-06-01:
> <https://github.com/prusa3d/Prusa-Firmware/issues/4697#issuecomment-2142672451>view
> it on GitHub, or

Thank you very much - I hadn't found this - now displaying Z-height as wanted.

James

At 18:14 31/05/2024, you wrote:-