prusa3d/Prusa-Firmware #4699: TX/RX pins on J19 : How to set up. What class to use?

RogerInHawaii, 2024-06-05:
The Einsy/Rambo board that Prusa firmware runs on has a J19 socket that includes TX and RX pins.

I'm working on a project that adds some extra hardware to the machine and includes an Arduino board attached to the TX/RX pins on the J19 socket. The Prusa firmware (modified by me) is intended to send text information to the Arduino through those pins on J19 with the Arduino responding by directly controlling my additional hardware.

I need to know what has to be set up or modified in the Prusa firmware itself in order for code in the Prusa Firmware to communicate (both TX and RX) with the Arduino via those TX/RX pins on J19.

I have tried dozens of approaches suggested by people in numerous forums, including the RepRap forum for the original Marlin firmware, and even ChatGPT and Copilot (I'm embarrassed to say) with no success.

Basically I need to know how to get the Prusa Firmware to communicate (with general text, not GCodes) over those TX/RC pins on J19 AND what class or object to use to do things like println().

I really hope someone can help. My project relies on this.

And, by the way, I'm not a newbie to programming. I'm a retired software engineer with 45 years experience, mostly C, C++, C# for PC application development.

RogerInHawaii, 2024-06-06:
I see that
 
Serial0
Serial1
Serial2
Serial3

and also

HardwareSerial0
HardwareSerial1
HardwareSerial2
HardwareSerial3

and even

SoftwareSerial

are things that have been mentioned to me as possible approaches, and even seem to be .cpp and .h files in the composite set of files for the firmware. But when I try them it results in compiler errors, most often saying duplicate definitions of things named like __vector_3.

PLEASE HELP ! ! !

3d-gussner, 2024-06-06:
@RogerInHawaii You have already an open isSuse https://github.com/prusa3d/Prusa-Firmware/issues/4698 please ask your questions there.
Closing it as it is a duplicate

closed 2024-06-06 by 3d-gussner