prusa3d/Prusa-Firmware #4700: [BUG] Z axes is going up after Unload filament is selected

slackista, 2024-06-08:
**Printer type** - [Any]
**Printer firmware version** -  [I think from the beginning]

Yet another Load/Unload issue. After press unload filament Z axess is going up +50.
* Harder to unload when using enclosure. No benefit with/without enclosure.
* After following failed unloads extruder reaches its E limit and still trying to going up. Impossible to fix clogged filament in enclosure.
* When last print finished eg at E=170 after pressing Unload then is completely failure, motoros trying to going up, very fard to unload filament in enclosure most probably resulting clog.

**To Reproduce**
Press "Unload filament" and wait until extruder is going +50

**Expected behavior**
After pressing Unload only setpoint heat should be applied and that's it. I see no reason to move extruder up to unload filament.

3d-gussner, 2024-06-25:
@slackista Thanks for the feedback. Lifting the Z axis while unload is to prevent the hot extruder staying close to the printed part and when filament is loaded also to be able to catch some oozing filament before moving back to the print.
The "Unload filament" procedure is used multiple times to reduce the code size.

One solution could be to only lift the Z axis a bit or not at all when the printer is IDLE. This would cost some precious resources and I am not sure that this will be implemented. I will discuss that with the other developers.

3d-gussner, 2024-08-14:
@slackista I have been trying to reproduce your request / issue and every thing is working correctly.
After the printer finishes at 170 the Z axis doesn't move 50mm up! As the printer "knows" its Z position it will not raise the Z axis above 50 mm.

Can it be that you turned off the printer (or reset it) with Z at 170mm and then later after turning it back on pressed Unload?

This would match your description as the printer shows `Z 0.00?`  on LCD and doesn't know the height after the reboot/start until you perform a Z homing.

slackista, 2024-08-14:
MK3 is sensorless homing, correct, use case is about after restart/reboot printer.

3d-gussner, 2024-08-14:
@slackista We changed it to 20mm BUT as the printer after reboot just can't know the height you may still have the issue when a print stops at 200mm and you turn off / turn on again followed by an Unload.

This issue has been closed automatically after merging #4750

PR #4750 (2024-08-14): Lower MIN_Z_FOR_UNLOAD
Lower the amount of Z lift on filament unload from 50mm to ~~10mm~~ 20mm.

This does not alter the behavior of M600. (M600 uses MIN_Z_FOR_SWAP)

Resolves #4700
closed 2024-08-14 by 3d-gussner