prusa3d/Prusa-Firmware #4701: [BUG]<Enter comprehensive title>

KDalen, 2024-06-11:
**Printer type** - mini
**Printer firmware version** -  6.0.1

**SD card or USB/Octoprint**
USB/Octoprint

**Describe the bug**
The heat test crashes the 3d printer

**To Reproduce**
 I ran the heat test

**Expected behavior**
  I would expect it to run the heat test

**Video**

https://github.com/prusa3d/Prusa-Firmware/assets/110197404/7975469a-8203-4cf8-b62f-9d5118e63b22

https://github.com/prusa3d/Prusa-Firmware/assets/110197404/66d39012-d089-48fd-8ef0-31ffb4b8a90c

  Please attach a video. It usually helps to solve the problem.

3d-gussner, 2024-06-25:
@KDalen This is the MK3/S/+, MK2.5/S repository, as you refer to the MINI and FW 6.0.1 please close this issue here and reopen it in the 32-bit MINI, MK3.5, MK3.9, MK4, XL repository https://github.com/prusa3d/Prusa-Firmware-Buddy

3d-gussner, 2024-07-08:
Wrong repository closing it.

closed 2024-07-08 by 3d-gussner