prusa3d/Prusa-Firmware #4702: [BUG]M< Error message at end of every Print>

DavidAIsom, 2024-06-11:
**Printer type** - [MK3.5]
**Printer firmware version** -  [6.02..]

**MMU upgrade** - [e.g. MMU3]
**MMU upgrade firmware version** - [3.0.3.]

**SD card is used for printing 
  

**Describe the bug**
  At the end of every print message appears 
[image]

**To Reproduce**
  Once a few filas ent loading errors have been made this message appears 

**Expected behavior**
  The only way to clear message on MK3.5 to clear message is to do a factory reset.  All will be ok until you get any more errors and5em it returns. With MK4 @cupajoe24 has found an option in the firmware to clear this counter/bit.

There's a reset in "Settings > Hardware > Nextruder Maintenance

We need a similar option for MK3.5
[image]

**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

MarkusReichart, 2024-06-14:
Opened in wrong repository. Should be moved to Prusa-Firmware-Buddy. Nevertheless I have the same problem. After fixing a loose filament sensor, this warning persists.

3d-gussner, 2024-06-25:
@DavidAIsom This is the MK3/S/+, MK2.5/S repository, as you refer to the MK3.5 and FW 6.0.2 please close this issue here and reopen it in the 32-bit MINI, MK3.5, MK3.9, MK4, XL repository https://github.com/prusa3d/Prusa-Firmware-Buddy

3d-gussner, 2024-07-08:
Wrong repository closing it.

closed 2024-07-08 by 3d-gussner