prusa3d/Prusa-Firmware #4704: [FEATURE REQUEST] Icon-based UI, taller buttons, or two columns for touchscreen menu items

reinux, 2024-06-22:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - MK4, XL
**MMU Upgrade** - MMU*

**Is your feature request related to a problem? Please describe.**
I find myself using the touch UI more and more often, but navigating the menus requires a degree of finger precision that I find a bit difficult to achieve consistently, even as someone with reasonable dexterity, having gone to college for music.

Some misclicks are quite time-consuming. For example, if you choose the wrong filament or nozzle, you have to wait for the hotend to finish heating up before the UI resumes.

**Describe the solution you'd like**
The buttons that appear during a print: Tune, Pause, Stop, are icons, and they work great. I wonder if the same could be offered for the menus, with a fallback to text-based when icons are missing. It looks like there's enough space to comfortably fit 4x2 buttons per screenful.

I can foresee several downsides to this in terms of development:
- Iterating becomes slower, as each button now requires a button. Perhaps a fallback to text-based UI could be a solution in case icons are missing.
- Typesetting could be more difficult, especially for languages that are already somewhat challenging, e.g. Japanese (although Japanese, with the help of kanji, is already more horizontally condensed than English). 

Another, simpler, solution might be to increase the vertical spacing between menu times. This is the approach that Windows took before the wider spacing became default in Windows 11 (or 10; I can't remember). The downside, of course, would be that the user would need to scroll more often.

Lastly, another solution might be to offer two columns as opposed to one full-width one. Button text could wrap around to take advantage of the extra spacing. The downside to this approach would be awkward line breaks, and potentially the need for hyphenation. However aside from the file menu, where entries can be as long as the filenames themselves, the Stuck Filament Sensor button is the only one that appears to be significantly longer than half the screen.

3d-gussner, 2024-06-25:
@reinux This is the MK3/S/+, MK2.5/S repository, as you refer to the MK4 and XL please close this issue here and reopen it in the 32-bit MINI, MK3.5, MK3.9, MK4, XL repository https://github.com/prusa3d/Prusa-Firmware-Buddy

reinux, 2024-06-25:
@3d-gussner Oh I see, sorry about that. Will do!

closed 2024-06-25 by reinux

3d-gussner, 2024-06-25:
@reinux No problem and thanks for the quick response and closing the request here, really appreciated :hugs: