prusa3d/Prusa-Firmware #4706: [BUG] Unable to resume print

alexiri, 2024-07-06:
**Printer type** - MK3S+
**Printer firmware version** -  3.14.0 (`MK3S_MK3S+-E3DREVO_FW_3.14.0_MULTILANG.hex`)

**MMU upgrade** - No
**MMU upgrade firmware version** - None

**SD card or USB/Octoprint**
  USB/Octoprint (Octoprint 1.10.2)

**Describe the bug**
  Unable to resume a paused print.

**To Reproduce**
1. Upgrade to 3.14.0
2. In PrusaSlicer, create a 10x10x0.8 cylinder shape
3. Insert a pause print in layer 0.6
4. Print the model
5. When the printer pauses, try to resume from the LCD panel. The `Resume print` menu item is missing.

**Expected behavior**
  While the printer is paused, `Resume print` is available and the print can be resumed. This works in version 3.13.3.

  The changelog for 3.14.0 mentions as a UX improvement that Resume "Is hidden while resuming" and then mentions that with host prints it's "only shown when the host is available". It's unclear to me if this means that Octoprint is missing functionality to notify its availability during resume (by sending `M79`?) If that's the case, then this breaks in backwards compatibility should be clearly stated.

**G-code**
[Pause-Bug_0.2mm_PLA_MK3S_1m.gcode.txt](https://github.com/user-attachments/files/16115655/Pause-Bug_0.2mm_PLA_MK3S_1m.gcode.txt)

**Photo**
[image]

alexiri, 2024-07-07:
I installed the plugin [Activate Prusa HostTimer](https://plugins.octoprint.org/plugins/ActivatePrusaHostTimer/) and now 3.14.0 works as expected and Resume print shows up. The changelog really needs to be updated to clarify that this breaking change requires a new Octoprint plugin!

3d-gussner, 2024-07-08:
@alexiri Thanks for the feedback.
The release notes point out that some host related menus are shown/hidden.

>   The host needs to send M79 to ensure that it can receive instructions from the printer.
>
>Resume
>
>   Is hidden while resuming
>   With host prints only shown when the host is available, see comment above.

>The community member @sarusani already implemented an OctoPrint plugin [Active Prusa HostTimer](https://plugins.octoprint.org/plugins/ActivatePrusaHostTimer/) 🎉

Why did we change that?

There have been complains that menus have been shown that "do not make sense" in some states of the printer. Some users "clicked" (accidentally or on purpose) and the printer behaved "strange" or prints even failed.
This has been fixed for the SD card prints but also apply for the host prints if the printer sends "Resume" via serial and no server is "listening".

Also to be able to introduce PowerPanic for PrusaLink / PrusaConnect host prints, we needed a solution to "know" when the host is back online to listen to commands send by the printer. The SBC or PC is always slower during boot up from a power failure.
Especially the "Resume" after PowerPanic is "critical" for host prints as the printer has to "know" that the host is back online and able to receive some information.

@sarusani published the OctoPrint Active Prusa HostTimer Plugin to use this new function also in OctoPrint.

Sorry for the inconvenience and hope that I could explain a bit why we "show/hide" some menus.

I will discuss your request to update the Release Notes.

Removed the bug label as it doesn't break the printer nor the PrusaLink/PrusaConnect usage. And with the `Active Prusa HostTimer` from @sarusani also OctoPrint is working correctly as you pointed out.

3d-gussner, 2024-07-26:
Closing this issue as solved

closed 2024-07-26 by 3d-gussner