prusa3d/Prusa-Firmware #4711: [QUESTION]How to enable Print area MBL

MrJoki007, 2024-07-10:
Please, before you create a new question, please make sure you searched in open and closed issues and couldn't find anything that matches.

**What is your question?**
Hello,
i have a MK3s+ with the newest firmware version 3.14.0 and want to enable Print area Mesh Bed Leveling, so it only levels the needed print area.
For that i enabled the 7x7 MBL.
Im using the newest version of prusa slicer 2.8.0 and did the newest configuration updates.

But it still levels the whole bed with 49 points.
How can i enable print area mesh bed leveling?
Do i have to change something in prusa slicer?

Thank you

3d-gussner, 2024-07-10:
@MrJoki007 
Please check that in PrusaSlicer -> Printer tab -> Custom G-code -> Start G-Code the line
`G80 X{first_layer_print_min[0]} Y{first_layer_print_min[1]} W{(first_layer_print_max[0]) - (first_layer_print_min[0])} H{(first_layer_print_max[1]) - (first_layer_print_min[1])} ; mesh bed levelling` exists.

Also if you have a "big" full size sliced part it will MBL all points of course.
When you open the gcode file with a text editor like notepad++ you should find a line starting with
`G80 ...`

Example:
`G80 X109.746 Y89.7458 W30.5083 H30.5083 ; mesh bed levelling`
`X109.746 Y89.7458` is the position of the front left corner of the part
`W30.5083 H30.5083` with a width of ~30 and height of ~30
The 25x25x2 "cube" has been centered on the Platter.

 
[image]

3d-gussner, 2024-07-24:
@MrJoki007 Closing this issue as it is working and thanks to @Max-Headroom pointing out that custom Prusa Slicer profiles need to be updated manually.

closed 2024-07-24 by 3d-gussner

leptun, 2024-09-23:
@EnduringGuerila 
You need to enable it on the printer as well. The setting is `Main menu -> Settings -> Mesh Bed Leveling -> Mesh [7x7]`. With newly sliced gcodes a reduced region is measured. With old gcodes the entire 7x7 is measured.

Don't forget to run a Z calibration as well from `Main menu -> Calibration -> Calibrate Z`, otherwise you might encounter issues during the probing. This is going to get addressed in the upcoming 3.14.1 where the printer will ask you to run Z calibration by itself when enabling 7x7.

EnduringGuerila, 2024-09-28:
Thank you, I did find this on the printer not terribly long after posting. I am surprised it wasn't documented better in Prusa's patch notes.