prusa3d/Prusa-Firmware #4712: fallo en perfil prusa slicer

jam2606, 2024-07-10:
### Description of the bug

al crear un archivo con prusa slicer 2.8.0 para mi mk3s+ com firmware 3.14.0 y mmu2s con firmware 3.0.2 y con el perfil de mk3s+ con mmu2 s ya sea en modo un filamento o multicolor al mandarlo a la impresora me dice que el archivo es para otra maquina

### Project file & How to reproduce

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.8.0

### Operating system

windows 11

### Printer model

mk3s+ con mmu2s

rtyr, 2024-07-10:
The g-code output uses correct printer model (for years), so it seems like some FW issue.
`M862.3 P "MK3SMMU2S" ; printer model check`

3d-gussner, 2024-07-10:
@jam2606 Thanks for reporting this issue.
I just tested FW 3.14.0+8066 (hash b97131bd) + MMU3 3.0.2 on MK3S+ with the [MK3S hex file](https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.14.0/MK3S_MK3S+_FW_3.14.0_MULTILANG.hex).

Also used PrusaSlicer 2.8 with newest profiles` Original Prusa MK3S & MK3S+ MMU3 Single` to sclice a 25x25x2 cube and sent it to the printer. No warning that the gcode is sliced for wrong printer.

Please ensure that you have flashed the printer with the `MK3S MK3S+` firmware and not the `MK3` hex file, which would explain the issue.

ojrik, 2024-07-10:
> ) + MMU3

MK3S+ FW 3.14.0 - MMU3 FW 3.0.2
It is no longer compatible with SLICER Profilom MK3S+ MMU2S. The printer will report an error in the G-code. For MMU from FW 3.0.2 Slicer Profile MK3S+ MMU3 should be used.

Firmware settings for MK3S+:
// Printer revision
#define PRINTER_TYPE PRINTER_MK3S
#define PRINTER_NAME PRINTER_MK3S_NAME
#define PRINTER_NAME_ALTERNATE PRINTER_MK3_NAME //the other similar printer to this.
#define PRINTER_MMU_TYPE PRINTER_MK3S_MMU3
#define PRINTER_MMU_NAME PRINTER_MK3S_MMU3_NAME
#define FILAMENT_SIZE "1_75mm_MK3S"
#define NOZZLE_TYPE "E3DREE"

// Printer name
#define CUSTOM_MENDEL_NAME "Prusa i3 MK3S+R"

XSSammy, 2024-07-12:
I have the same issue same printer with a mmu2s.  @ojrik I haven't tried printing with the mmu3 profile but I can confirm it doesn't report the same error if I use that profile.

ojrik, 2024-07-12:
FW from 3.0.0 to 3.0.3 is for MMU3 device. If you use the mmu2s profile, the G-Code will be for a different type of printer. I have MK3.5 MMU3, MK3S+ MMU2S, MK3S MMU3 Ulti multi everywhere I use SLICER profile for MMU3. By the way, I made g-cod for MK3S+ MMU2S and started printing on my MK3S+ FW 3.14.0 - MMU2S FW 3.0.2 device reported G-Code error for different printer type.

@3d-gussner  also used a profile for MMU3 and no report.

@XSSammy do you have a similar but different report?

ojrik, 2024-07-12:
A small note, if you just flashed FW 3.0.2 to the MMU2S device without UPGRADE, you already have MMU3.

closed 2024-10-15 by github-actions[bot]