prusa3d/Prusa-Firmware #4714: [BUG] Some gcodes cause MK3S to reset, give z-axis calibration error, and lose all settings forcing complete re-calibration

zaniyah, 2024-07-13:
This is a Prusa MK3S, with the latest firmware (3.14) printing from the SD card.

Every time I try to print a specific gcode file, the printer fails saying that there has been a z-axis error.  At first I thought that this was a genuine hardware issue, and after tweaking and recalibrating and having successful prints of other things I had printed before, I tried again.

This cycle has been repeated several times now, and the same gcode file always triggers this bug.

Download the top part of this box here as a gcode file:

https://www.printables.com/model/3376-e3d-v6-nozzle-holder-nozzle-holder-nozzle-case/files

Specifically, this one:

https://files.printables.com/media/prints/3376/gcodes/21562_24bde222-9a5d-4c5e-9dd8-2169feb70d7d/top-piece-nozzle-holder_01mm_pla_mk3.gcode

Put on the SD card and attempt to print with a Prusa i3 MK3S with version 3.14 of the firmware, 0.4mm standard brass nozzle and PLA.

Initial bed calibration is fine.  When printer starts to print, it almost immediately resets and gives a z-axis error.

rspigner59, 2024-07-13:
I'm having the same issue, MK3S with firmware (3.14) printing from SD card. About every fourth print the print re-calibrates the Z-axis.  Same info as mine: 0.4mm standard brass nozzle and PLA. 
Same error here as well: Initial bed calibration is fine. When printer starts to print, it almost immediately resets and gives a z-axis error

gudnimg, 2024-07-13:
Hi 👋 I tried to reproduce this on my MK3S+ but so far have not been successful.

It would help us if you could share exactly what the z-axis error says?

Also when you say "reset" are we talking about a machine reboot or some state of the printer FW?

If you could share the serial logs from the printer that would also be super helpful. The baudrate is 115200.

zaniyah, 2024-07-15:
I don't think you'll be able to recreate it with a MK3S+.  I have been doing some fiddling around (I now have the upgrade kit) and it seems that under certain circumstances the firmware mistakes the normal older pinda sensor reading for an invalid one, triggering the re-calibration of the z-axis.

rspigner59, 2024-07-15:
I have an MK3S printer and bought the MK3S+ upgrade kit. Just haven't installed it yet.

zaniyah, 2024-07-19:
"Some error encountered, Z-levelling enforced" is the intermittent error.  The first error we saw was a completely corrupted screen, and the firmware crashed hard.  I had to re-flash the printer.

Now I have this error intermittently, which is making it really, really hard to print the required parts for the MK3S+ upgrade.  There seem to be many default settings that are for the MK3S+ and MK4 SuperPINDA but not appropriate for the older MK3S.

rspigner59, 2024-08-01:
There needs to be a fix in the firmware for the MK3S to stop the automatic Z-axis calibration from taking place.

It's not good for the stepper motors since the MK3S lack up-limits to prevent the stepper motors from attempting to move the print head beyond the end of travel. This occurs about every 3rd or 4th print, after bed leveling, then the printer performs a z-axis calibration before printing the object.

Emmanuel2102, 2024-08-13:
Did you activate the 7x7 Z levelling new feature ? I assume that this is this one that doesn't work well. I have solved the issue by setting back the 3x3 bed mesh levelling.

zzeng5, 2024-08-13:
Had same issue on my MK3s 3.14 firmware with 7x7 mesh leveling enabled.

rspigner59, 2024-08-13:
"Did you activate the 7x7 Z leveling new feature ?"
No, mine is [3x3] . The only thing I have done is upgrade to firmware (3.14). Prior to f3.14, I don't recall ever having a z-axis calibration error.

Scandiravian, 2024-08-15:
Seeing the same issue on my mk2.5s. I've never had an z-axis error in the 7 years I've had the printer, but after upgrading to 3.14 I have to run recalibration after most prints

3d-gussner, 2024-08-30:
@Scandiravian Did you run the Calibrate Z? This should fix the issue.

Scandiravian, 2024-08-30:
@3d-gussner thanks for the suggestion, but unfortunately that fails as well. I have to run a full XYZ-calibration and change z-calibration to 3x3. All other combinations fail

3d-gussner, 2024-09-05:
This [pull request](https://github.com/prusa3d/Prusa-Firmware/pull/4765) should fix the issue which will be in next release FW 3.14.1.

harriettubgirl, 2024-12-03:
I keep having this happen also, 3.14 with MK3S+, never had this until I upgraded firmware.

3d-gussner, 2024-12-03:
@harriettubgirl Sorry to hear that, please update the firmware to [3.14.1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.14.1)

harriettubgirl, 2024-12-03:
> @harriettubgirl Sorry to hear that, please update the firmware to [3.14.1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.14.1)

Oh thank you, I didn't even see that was out heh

closed 2025-02-22 by Prusa-Support

wdominguez2011, 2025-04-29:
I just had this same error happen when trying to print a specific file after just printing successfully two days ago. It’s my 3rd print so I’m still very much a noob. I know that nothing moved or changed from the last print so I was thrown off by this need to recalibrate. After an hour of trying some things, I just decided to start the print job over again, and it worked.

Have you guys just tried to restart the print job when seeing this error? It’s been going for an hour or so, and the print looks great. It may be an issue with the g-code but I tried the old “unplug and try again” method and it calibrated everything at the start of the job and it’s going well. It may just be a bug that requires starting the print job over again for some or maybe most..