prusa3d/Prusa-Firmware #4716: [FEATURE REQUEST] Don't cool down nozzle right away on print pause (M601)

alexiri, 2024-07-15:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - all
**MMU Upgrade** - N/A

**Is your feature request related to a problem? Please describe.**
  I'm always frustrated when I pause a print (to add a nut, magnet, etc.) and the nozzle immediately starts to cool down. When resuming a few seconds later, I have to wait for the nozzle to warm back up.

**Describe the solution you'd like**
  After pausing, keep the nozzle hot for a few minutes, in case the user resumes the print quickly enough.

3d-gussner, 2024-07-15:
@alexiri Thanks for the request, we had similar requests before and there are two sides on that.
1. If you stand next to the printer when it is going to pause, then I agree that the nozzle cool down could be delayed
2. But often you have the pause in the gcode and don't babysit the printer, which means that you come back after some time and then the filament is "cooking".

So the question is what is a good way to improve it for point 1 while preventing the filament being cooked and degrade.

I assign this request to myself and discuss that with the team and filament department.

Can't promise anything, let's see if and what we can do.

alexiri, 2024-07-15:
How long does the filament take to "cook"? Waiting 60 seconds before cooling might already be good enough.

3d-gussner, 2024-08-08:
@alexiri Just an update: I got some feedback about the cooking. After looking at the PrusaXL Prusa Slicer Profiles most Idle temperatures, where the tool is waiting, are around 70 -100 °C.
One feedback was that "some" filaments tend to overcook if not extruded but couldn't get any specific values.

We will discuss the idea to give the pause a short time period before cooling down. Maybe set some Idle temperatures similar to the XL profiles to reduce the heat-up time from cold/idle to print temperatures during pauses.

vincele, 2024-11-11:
Maybe only do a partial cooldown to avoid filament cooking, but stay warm enough to be able to start printing again quickly.