prusa3d/Prusa-Firmware #4718: Unload Filament Command Only Gives 30 Seconds to Comply

nathan118118, 2024-07-16:
In previous versions of the firmware, choosing "Unload Filament" on an MK3s+, and then the filament type, would start a pre-heating process. Once it was up to the necessary temperature, the printer would sit and wait on a screen that said something to the effect of "click to unload." And it would just sit there for a LONG time. Eventually it would time out...after maybe close to 10 minutes, and give the heating disabled due to safety timer.

In recent changes to the firmware, now when you click the filament type to unload, it again heats up to temperature, but then gives only 30 seconds to click to unload. If you don't click it in that small window, it goes back to the main menu, forcing you to scroll and click "Unload Filament" again. I timed it, and I think it was 30 seconds exactly, so it seems like this value was changed intentionally.

This change makes very little sense. If you only have a single printer, that means you have to stand there and watch it. You can't work on anything else, change filament colors, etc. This problem is even worse if you run a print farm. Previously I would tell all my printers to unload, then I could do all my filament color prepping, and then each printer would be waiting asking me to click to unload. I could go through all of them so fast. Click click click click and unload a bunch of them.

Now with this 30 second timeout, I have to scroll and click unload on every single printer. I know that doesn't sound awful, but why the 30 second timeout at all? If a user wants to access anything else like changing sheet type, they can do that before or after the filament unload.

Love PrusaSlicer, but this specific change has been a step backwards for me personally. I don't know if this change was introduced with the 3.13 series of firmware, but I'd love for it to go back to sitting and waiting for the user. After all...if the user clicks unload filament, that's what they want to do! 🙂

gudnimg, 2024-07-25:
Hi 👋 I suspect this change introduced the regression: https://github.com/prusa3d/Prusa-Firmware/commit/a7e9ccfb57e436311cb28b5636b8f66fe7982909#diff-5fbfdca43c192573e31d8e78d255c510fb6e213b118cf38173277ec2a2163880R7462

It was introduced in 3.13.0 when we did some major refactoring to consistently consume knob clicks in the menus, if I'm correct this was completely unintentional. Thankfully the fix is trivial: we need to explicitly tell that menu to not timeout.

[image]

nathan118118, 2024-07-25:
> Hi 👋 I suspect this change introduced the regression: [a7e9ccf#diff-5fbfdca43c192573e31d8e78d255c510fb6e213b118cf38173277ec2a2163880R7462](https://github.com/prusa3d/Prusa-Firmware/commit/a7e9ccfb57e436311cb28b5636b8f66fe7982909#diff-5fbfdca43c192573e31d8e78d255c510fb6e213b118cf38173277ec2a2163880R7462)
> 
> It was introduced in 3.13.0 when we did some major refactoring to consistently consume knob clicks in the menus, if I'm correct this was completely unintentional. Thankfully the fix is trivial: we need to explicitly tell that menu to not timeout.
> 
> ![image](https://private-user-images.githubusercontent.com/8218499/352238138-6a29520a-099b-4ab9-b103-9b39f16afb09.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE5MzI2MzIsIm5iZiI6MTcyMTkzMjMzMiwicGF0aCI6Ii84MjE4NDk5LzM1MjIzODEzOC02YTI5NTIwYS0wOTliLTRhYjktYjEwMy05YjM5ZjE2YWZiMDkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MjVUMTgzMjEyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODEyZjM2NTg3ODcwN2U4NThkYjE3YjM4YmVjOTFhYzQ0OTljMTlmYTRlYTRhZTJhZjg4Y2ZlY2YyYTBhZDIwNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.NmPd0I27vgMW6t48Vf8dzNIyaHySZ9TT3_njyUPhqD8)

Thank you so much for taking a look at this! Really appreciate it. :)

closed 2024-07-26 by 3d-gussner