prusa3d/Prusa-Firmware #4720: [FEATURE REQUEST] Printer lockout

nyckidryan, 2024-07-17:
I work with a maker space and we have a farm of 13 printers, 8 Minis, 4 MK4s, and an XL. We've been having some issues with untrained members using the machines. I've looked through forums and pinout tables trying to find something that stuck out, but I haven't come up with anything. While custom firmware for the machines would be an option, I believe this would be useful for other maker spaces, libraries and schools.

I'm looking for a way to lock out a printer from printing anything until an external authorization is presented, possibly by way of a GPIO pin or a serial command. Similar to a copier in an office that requires an account number before operating.

The user would be authorized by way of a device next to each printer that would read a member's card / PIN / key switch and if authorized, close a relay connecting a GPIO pin to VCC / GND, or perhaps by sending some gcode or other command through the serial port to unlock the machine. A simple auth check at the beginning of the print sequence would be sufficient - simply ignore the print request. Or for more feedback, return a "Not authorized" / "Printer not ready" message - perhaps something that could be set by gcode or PrusaLink and stored in eeprom.

There is a system that exists from [Fabman](https://shop.fabman.io/products/fabman-bridge-v2) which uses a member's card to enable a device, however it works by connecting and disconnecting power to the device, and monitoring for a low current draw to indicate the is done and forgot to log out. Great for equipment like soldering irons, table saws, drill presses, and this would work for the most part, but we don't like to just pull the power from a printer, especially if the access box would potentially interpret a printer waiting for a filament change as a completed print and just shutting down.

3d-gussner, 2024-07-17:
@nyckidryan This is the repository for 8-bit MK2.5/S and MK3/S/+ printers please close this issue here and re-open it in the 32-bit repository for MINI, MK3.5, MK3.9, MK4 and XL https://github.com/prusa3d/Prusa-Firmware-Buddy

nyckidryan, 2024-07-18:
@3d-gussner Thanks for the heads up ... don't git while drowsy! 😄

closed 2024-07-18 by nyckidryan

3d-gussner, 2024-07-18:
@nyckidryan No problem, I have one or two ideas for the request you have but want to answer these in the correct repository so the 32-developers and community can see it.