prusa3d/Prusa-Firmware #4721: [BUG] Thermal Anomoly errors in 3.14.0

ezekeal, 2024-07-17:
**Printer type** - MK3S+
**Printer firmware version** -  3.14.0

**MMU upgrade** - none
**MMU upgrade firmware version** - none

**SD card or USB/Octoprint**
  sd card

**Describe the bug**
  After updating to firmware 3.14.0 I am getting thermal anomaly errors during calibration.
  Previously the printer was working without issue.
  All stock parts no upgrades

**To Reproduce**
  Walk through calibration wizard
  anomoly happens on first layer calibration

**Expected behavior**
  No thermal anomaly errors

ezekeal, 2024-07-17:
It's clear from the forums that this thermal model code has never worked well. What is really needed is a way to disable it.

Swejzi, 2024-08-01:
I have same problem :( 

I was upgrading my MK3S+ from 3.13.2 to 3.14.0 and I got this error. When I downgrade from 3.14.0 to 3.13.3 it works fine without this error

donslice, 2024-08-21:
+1 for the thermal anomaly here.
MK3S, newly upgraded to 3.1.04 today. It was previously working with no issues before upgrading.

bradtho, 2024-09-07:
I have replicated the issue on my MK3S+. I have a [Revo Six Hotend](https://e3d-online.com/products/revo-prusa-coldside-kits) 

My last working version of Firmware is 3.11.0. The issue was introduced with 3.12.0 with the addition of the "New thermal model protection" feature.

I recall there being a discussion to this effect for those who don't use the stock extruder.

GaborTheHutt, 2024-09-07:
Same problem here.

dingo1984, 2024-09-14:
i did buy my machine second hand had no issue does have a nozzle upgrade. soon as i upgraded to 3.14.0 i've had the thermal anomaly. and wont calibrate

GaborTheHutt, 2024-09-16:
I did a factory reset after and before flashing 3.14 and then the thermal anomaly error appeared immediately.
Gone back to 3.11 for now.

cpuks, 2024-09-18:
Same here on mk3s+ with revo and bondtech extruder, sometimes thermal error persist but hotend warms up without issues, sometimes with same error on display warm up stops and I have to clear error.

bradtho, 2024-09-19:
@GaborTheHutt, @dingo1984 and @cpuks - Are you using the standard 3.14 image or the special [Revo 3.14](https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.14.0/E3D_REVO_FW_MK3_MK3S_MK3S+_FW_3.14.0.zip) image? The discussion in #4451 goes into detail on how to install it.

cpuks, 2024-09-19:
Since I've got bondtech extruder I'm using their version and it's special one for revo - and it's displayed on the screen. So well I think I'll have to go back to 3.11 as turning off thermal model does not work in my case.

3d-gussner, 2024-09-20:
> as turning off thermal model does not work in my case.
@cpuks What do you mean that turning off Thermal model doesn't work?

You can connect via USB and open a serial monitor and look which TM errors you get.
Can you share your log file of the issue you describe?

I get sometime TM warnings when heating up after a print has finished and I start a new print, BUT never got a TM error that I had to clear. On the other hand I have the stock E3Dv6.

Try to change your `W` and `E`  values to bit higher values to prevent TM errors that need to be cleared.
Please do not forget that changing the settings with `M310` are only active until you reset or power off the printer.
To save your settings permanent you need to use `M500`.

I can understand some user being frustrated that TM triggers on some REVO to sensitive and what to disable it. But there have been more than one case I know off where a broken wire was detected and so prevented a way bigger issue.

dingo1984, 2024-09-22:
i not sure as i didnt ask when i bought it i have flashed again to 3.14 still getting the anomaly i clear  tm fault and retry first layer calibration. i do this several time until i reach around 160c alarms but keep going to right temp but now filament sensor not working

CaptainFalcon92, 2024-11-16:
The thermal model is completely broken.
I've been delaying over and over my upgrade to a newer firmware, I used to run 3.11 until only a few days ago. The printer is now stupidity annoying.
As soon as I upgraded, jumping to 3.14 : thermal anomaly. Beep beep thermal anomaly ! Right now as I'm writing these exact words, the printer has finished a job and is just cooling down. It beeps ! All the time !
The printer yells at me during filament change, during intentional pausing, when the fan ramps up, or simply during preheats. Prusa have been blaming the HW for a couple years, still everyone knows the hardware is perfectly fine. 3.11 proves it. 
This TM protection model thing must be turned completely off for Revo. It simply is not required. Besides, years of previous firmware did not required it either. Truth is nobody asked for it. I would love to simply have an option to kill that thing from within the printer settings menu.

3d-gussner, 2024-11-16:
@CaptainFalcon92 

> This TM protection model thing must be turned completely off for Revo. It simply is not required. 

You can disable the TM using the [M310](https://prusa3d.github.io/Prusa-Firmware-Doc/Prusa-Firmware/group__GCodes.html)  gcode command and to make it permanent use `M500`.

>Prusa have been blaming the HW for a couple years, still everyone knows the hardware is perfectly fine. 3.11 proves it.

I am aware of at least one REVO user who has found a broken wire at some specific position as the connector rubbed off the isolation. I would argue that it shows that the connectors of the REVO can cause issues and the hardware isn't "perfectly fine".

> Besides, years of previous firmware did not required it either.

The TM works well for the E3D v6 stock nozzle.
TM has detected multiple broken wires and thermistors. 

The REVO values have been provided by a small community group and E3D and so these values may not work for every one.

>Right now as I'm writing these exact words, the printer has finished a job and is just cooling down. It beeps ! All the time !

Did you try the TM calibration to fine tune the TM values to your specific printer?

>The fact there is a "unable to reproduce" tag on this issue is borderline mocking us.

The REVO is a 3rd party solution and issues should be reported to E3D.
We try to help everyone but sometimes we (which have stock Prusa printers) can't reproduce these, especially when 3rd party parts are used and/or the printer is modded.

> I would love to simply have an option to kill that thing from within the printer settings menu.

There will be no menu to enable or disable this feature.

E3D is responsible for the support of the REVO! As E3D is a long partner of Prusa we agreed to compile and release the firmware here for E3D.

Reading these kind of comments let me think that we may have to stop compiling and releasing the firmware for the REVO and let the community / E3D to do it themself.

mhouse1, 2024-11-23:
i made an update on https://github.com/prusa3d/Prusa-Firmware/issues/4451#issuecomment-2495462771

the gist of it is we're reverting back to fw3.11

mhouse1, 2024-11-23:
> The TM works well for the E3D v6 stock nozzle

that may be but there should be no reason to run the V6 nozzle with revo available considering how V6 is to prone to leaking and maintenance is time consuming

3d-gussner, 2024-11-25:
>there should be no reason to run the V6 nozzle with revo available

The original MK3S comes with an E3Dv6. There might be some advantages to use the REVO but Prusa doesn't sell these.

closed 2025-02-04 by github-actions[bot]