prusa3d/Prusa-Firmware #4722: No 'Load Filament' menu option when broken filament sensor always 'senses' filament

dzuberbier, 2024-07-18:
**Printer type** - MK3S+
**Printer firmware version** -  3.14.0

**Describe the bug**
  If the filament sensor is malfunctioning and always detects filament (Support > Sensor Info > 1), the 3.14 firmware will not show a 'Load Filament' option on the LCD menu due to the UX improvements listed in the 3.14 release notes. Re-flashed to 3.13 for 'Load Filament' to be a permanent option in the Menu.

**To Reproduce**
  Must have malfunctioning filament sensor that always detects filament.

**Expected behavior**
  I expected 'Load Filament' to always be an option in the LCD menu as in previous firmware versions.

ojrik, 2024-07-18:
Greetings. If you have a broken filament sensor, just turn off the filament sensor in the settings (Settings-Fil.Sensor-Fil.Sensor [Off]). LOAD FILAMENT will also be displayed in the LCD menu.

dzuberbier, 2024-07-19:
Thank you for stating the obvious and I'm disappointed in myself for not thinking of it myself. 

I tried turning off the filament sensor in 3.13 and the machine would not let me turn it off. I reflashed 3.14 and the machine still wouldn't let me disable the filament sensor. https://www.dropbox.com/scl/fi/6n4t95qrw0m0awbly6xbe/MVI_1435.MP4?rlkey=e0yys3590d73jgnz1nsvlg77p&st=m6ld50c7&dl=0

Only after unplugging the connection from the sensor itself did 'Load Filament appear in the menu. Unfortunately, that means the 'Unload Filament' option doesn't appear.

ojrik, 2024-07-19:
greetings. in the video you can see how you turn off and immediately turn on the filament sensor. If fil. the sensor is in the off state, do not click on the file anymore. you have already switched off the sensor. Just click on the main menu.

dzuberbier, 2024-07-19:
Oh boy...I'm having a rougher week than I thought. My brain thought I was going one step deeper into the menu. Thank you for your help...

closed 2024-07-19 by dzuberbier