prusa3d/Prusa-Firmware #4723: [BUG] Z Axis calibration errors during G80 on 3.14 until a Calibration>Z Calibration is run

tideline3d, 2024-07-22:
After upgrading our 36 MK3S+ printer farm to 3.14 yesterday, I started ~20 printers today and about 8 of them failed to start their next print with a Z-axis calibration error.  When I investigated, I saw that the plates were on fine (usually the cause of this).   Allowing the printer to home the Z up top and retry the G80 would not solve the issue, even with a couple printer resets.  Most of them failed on the first 10 or so points, but one or two of them made it all the way to the back line before failing, again all with properly installed plates. 

_**I've managed to resolve each of these by running Calibration > Z Calibration**_.    All 8 printers resolved their symptoms immediately after running the Z Cal, but interesting that the automatic Z calibration that's kicked of when G80 finds a problem does NOT solve the problem, it had to be from the Calibration menu.  

These GCodes mostly came from 2.7.4 PrusaSlicer and do have the parameters for G80 showing the bounding box of the print.   My printers all have 7x7 MBL turned on.  They are all running OctoPrint from a RPI4 with a plugin sending M79 pings every 20s to the printer so we can utilize the 'Set Ready' (thanks for this!) feature for our internal print job mgmt system

These GCodes are the same ones we print every day and haven't been re-sliced in a few months.   

Here's the start gcode for one of the prints that was acting up.   Note that Y is a negative value, is this expected?   This print takes up a large portion of the plate and has a skirt that comes near the limits of Y.  

[image]

```  
  M862.3 P "MK3S" ; printer model check
  M862.1 P0.8 ; nozzle diameter check
  G90 ; use absolute coordinates
  M83 ; extruder relative mode
  M104 S170 ; warm nozzle to idle
  M140 S90 ; set bed temp
  M190 S90 ; wait for bed temp
  M109 S170 ; wait for extruder idle temp
  G28 W ; home all without mesh bed level
  G80 X22.0967 Y-0.203449 W205.807 H210.407 ; mesh bed levelling
  G1 Z2 F720 ; raise Z so ooze doesnt bind to nozzle
  M104 S240 ; set extruder temp
  M109 S240 ; wait for extruder temp
```

3d-gussner, 2024-08-30:
@tideline3d Please consider to close this issue as the Z calibration fixes these issues.

goldrake1981, 2024-09-12:
I also have the same problem since I installed 3.14. You say that by calibrating Z from the Calibration menu solves the problem? In my case it is not so.

rtarb, 2024-09-12:
I also have this issue. Running z calibration does not fix the issue. Wasn't a problem until I updated.

Hesse56, 2024-11-14:
Same : "I also have this issue. Running z calibration does not fix the issue."

tideline3d, 2024-11-14:
It looks like they may have addressed this with #4781 in the 3.14.1 branch.   It was definitely new to 3.14.0 for us but we've noticed that the printers that caused this issue had some beds that were quite skewed and have since shimmed the beds and don't have this error any more.

3d-gussner, 2024-11-28:
Please check https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.14.1.
Closing this issue.

closed 2024-11-28 by 3d-gussner