prusa3d/Prusa-Firmware #4724: [FEATURE REQUEST] After 'Set Ready', go back to main info screen

tideline3d, 2024-07-22:
The new 'Set Ready' feature works great, but it's awkward that it stays in the menu after clicking it.  Then you can't see if a new job was picked up via the LCD message or by watching the temp targets change.

Would be great if after clicking 'Set Ready' you're exited out of the menu and put back to the main info screen.

sarusani, 2024-07-26:
Am I assuming correctly, that you use it in combination with with OctoPrint? The current behavior is optimized for use with PrusaLink.

tideline3d, 2024-07-26:
Yep, with OctoPrint.  I looked through your plugin and mimicked into my own.  How does PrusaLink handle that differently?   When I receive the action:ready command, I start a print in OctoPrint quite similar to yours.   I'd just like the printer LCD to go back to the main screen when that happens.    The only way I can see it started a print is if the 'Tune' menu pops up, I'd rather see my M117 messages come through on the main screen to confirm the print is starting. 

  I assumed this was a firmware thing, but if there's something I can send back from the host to have the LCD go back to the main screen, that'd be just as good from my side.

sarusani, 2024-07-26:
PrusaLink uses "Set Ready" a bit different and answers with a `M72 S1`. This will change the LCD menu to show  "Set not Ready". So it makes sense to stay in the Menu. (So you can toggle between the 2 states)
This made no sense in OctoPrint, so my plugin doesn't send the M72...

Currently I don't have a proper solution for you.

In my opinion, when the host is OctoPrint, the best solution would be to replace the "Set Ready" menu entry completely with an "Start Host Print" entry that sends "action:start" (natively supported by OctoPrint) instead of "action:ready". (And then of course closes the menu after it's been selected.)

Maybe that's something that could be discussed for a future release.

tideline3d, 2024-07-26:
Fair enough, it still works for me, I may just fork the Firmware if it bothers me that much.   FWIW it still does toggle to 'Set Not Ready' even though I'm not sending back M72 S1.  

In our homemade system we'll put M117 messages onto the screen if a) there's no job waiting or b) the filament listed doesn't match the next job.   We dont actually load them up in Octoprint, the set ready goes out and polls for jobs waiting for that printer and starts one if all the conditions are right.   Prior to 'Set Ready' we just used the 'PETG Preheat' as our trigger.  If a user preheated to PETG temps and there was a job waiting, we'd start it.   Set Ready is definitely an improvement.    

We also can't use 'Reprint' because we track every print job that's managed across the farm.   It's been a bit of an issue accidentally hitting that since I have thousands of reps of muscle memory with the 2nd position being 'Preheat'.   

I assume if I wanted to fork it and the menu close after Set Ready, I could just add   lcd_return_to_status(); to lcd_printer_ready_state_toggle over in ultralcd.cpp?

sarusani, 2024-07-26:
Yes, lcd_return_to_status() should do the trick.

closed 2024-10-03 by github-actions[bot]