prusa3d/Prusa-Firmware #4725: [BUG]MK3S doesn't load filament

majedation, 2024-07-23:
**Printer type** - MK3S
**Printer firmware version** -  3.14.0

**SD card or USB/Octoprint**
  SD Card

**Describe the bug**
  The printer doesn't load the filament. The nozzle is preheated, the filament sensor recognizes that the filament is inserted, and the autoload is active, but the printer doesn't load the filament. This happened directly after unloading a filament.
  After resetting the printer through the reset button and preheating again, it loads successfully.

**To Reproduce**
  Unload the filament, then reinsert the filament.

**Expected behavior**
  Should load successfully

gudnimg, 2024-07-27:
I am unable to reproduce this on MK3S+. Perhaps this is a MK3S specific bug? I don't have a MK3S machine to test on unfortunately.

My MK3S+ uses the MMU filament sensor setup, these are the steps I used to test:
1. Turn on the printer
2. Disable MMU in settings
3. Make sure AutoLoad in enabled
4. Push the lever on the filament sensor chimney to trick the sensor to detect filament => Autoload is triggered an I am prompted to preheat
5. Wait for the preheat to complete and reach 215°C
6. Click 'Yes' to confirm the filament was loaded correctly
7. Go to Settings -> Unload Filament
8. Repeat Steps 1 - 6 (I see no issues here). The only difference is Preheating is skipped since the nozzle is already preheated.

majedation, 2024-07-28:
It doesn't happen to me every time. I think the last time it happened to me was right after the end of a print. I unloaded the previous filament and inserted the new one, and when I inserted the new one, this issue appeared.
I can re-update you when it happens to me again; I'll try to remember in what sequence this bug occurs.

gudnimg, 2024-07-28:
@majedation thanks for the reply, I wonder if perhaps the could be the sensor reading flickering? There could be some timing differences between the new firmware and the older firmware that could cause more or few sensor samples to be read. I'm just speculating of course so take this with a grain of salt.

closed 2024-10-08 by github-actions[bot]