prusa3d/Prusa-Firmware #4726: [BUG] Issue with mid print color change

majedation, 2024-07-23:
**Printer type** - MK3S
**Printer firmware version** -  3.14.0

**SD card or USB/Octoprint**
  SD Card

**Describe the bug**
  I've inserted a color change in the slicer in the middle of the print directly using the "+" in the slider, not through custom g-code. When the printer reaches the color change step, the printer stops and parks the nozzle to the right and beeps to unload the filament; then I press the knob to unload the filament, then press the knob again "Yes" to confirm that the filament is removed.
Everything runs till here successfully; there are no issues here.

Issue: When I insert the new filament, it doesn't load, and then when I remove it, the extruder starts loading the filament.
The extruder starts loading the filament after the filament sensor senses the removal of the filament, not insertion; it should be the opposite.

I'm using PrusaSlicer 2.8.0.

**To Reproduce**
  Insert filament change in the slicer using the "+" on the right slider.

**Expected behavior**
  The filament should be loaded after inserting the filament.

gudnimg, 2024-07-27:
Isn't this the same issue as in https://github.com/prusa3d/Prusa-Firmware/issues/4725? I don't think the problem is M600. If AutoLoading doesn't work from the menus, then it won't work with M600.

majedation, 2024-07-28:
It's a different one. Issue #4725 occurs while the printer is idle, but this one occurs during mid-print.
Auto loading is active. The extruder loads the filament automatically, but when the filament sensor triggers the filament is removed.
What happens is that when the printer asks me to load the new filament mid-print after unloading the previous one, I insert the new filament; nothing happens. When I remove the filament, it starts loading, without pressing the knob.
It happened to me multiple times.

gudnimg, 2024-07-28:
The underlying code is the same though from Auto loading feature's perspective. The only difference with M600 is it is doing a lot more UI stuff.

> When I remove the filament, it starts loading, without pressing the knob.

In M600, the image below shows the relevant code. The 'loading' you describe is simply automatic movement which is meant to 'finish' the loading. `fsensor.getFilamentLoadEvent()` is returning `true` and so thinks you've inserted filament. The fact the printer thinks it's seeing filament when you remove it sounds like the sensor reading is flickering 🤔 

There's unfortunately not much I can do to help unless this can be reproduced reliably.

[image]

majedation, 2024-08-05:
This happens to me every time. I've recorded a video, but the file size is large. How can I add the video here to this issue?

gudnimg, 2024-08-05:
If the file is too large then I would personally try either:
1. Create an unlisted video on Youtube (only shareable via link)
2. Share the video via Cloud storage like Google Drive such that anyone _with_ the link can view it.

gudnimg, 2024-08-05:
Have you tried:
1. Disabling Autoload
2. Go to Menu > Support > Sensor Info > Fil. sensor
3. Stick a piece of filament into the extruder (does is read steady 1?)
4. Repeat a few times. Try also moving the filament piece side-to-side or insert it at an angle.

@Prusa-Support could you take a look at the video? Any thoughts?

closed 2024-10-15 by github-actions[bot]