prusa3d/Prusa-Firmware #4729: [BUG]<Enter comprehensive title>Power Panic Recovery not working

Glennk60, 2024-07-26:
**Printer type** - MK3S+
**Printer firmware version** -  3.14.0

**MMU upgrade** - None
**MMU upgrade firmware version** - N/A

**SD card or USB/Octoprint**
  Print via SD card

**Describe the bug**
  I have 6 MK3S+ printers, that I updated to firmware version 3.14.0  And I have had very short power outages, maybe only 2 - 5 seconds, just enough to shut down appliance clocks, the printers, modems, etc. But none of the printers have recovered since installing the update. I was excited to hear the new version had "Improved Power Panic" management, but that hasn't been the case with mine. Power Panic DID work on previous versions.

gudnimg, 2024-07-26:
Hi 👋 Can you describe a bit more what didn't work? Was it a problem with LCD user interface? layer shift after recovery? no recovery at all perhaps? Anything else which specifically didn't work?

Anything which could tell us the point of failure would be helpful.

Glennk60, 2024-07-26:
Hi, there was no recovery at all, on any of the machines. The LCD screen returned to "normal", just as it is when you turn it on, or when you push the reset button. No ability to resume print. The print head remained over the current object being printed. It literally stopped where it was, and reset.

gudnimg, 2024-07-26:
> Hi, there was no recovery at all, on any of the machines. The LCD screen returned to "normal", just as it is when you turn it on, or when you push the reset button. No ability to resume print. The [...]

Hmm that sounds like the printer didn't complete/finish the process of saving all the data 🤔 The last thing the firmware does is write to EEPROM so that the next boot-up knows a power panic happened, and after that it will move the extruder to the side (X-axis).

closed 2024-10-15 by github-actions[bot]