prusa3d/Prusa-Firmware #4734: [BUG] Display remaining time for color change on the LCD screen

majedation, 2024-07-28:
**Printer type** - MK3S
**Printer firmware version** -  3.14.0

**SD card or USB/Octoprint**
  SD Card

**Describe the bug**
  In the previous firmwares, the MK3S showed the remaining time for color change; it toggled every few seconds between the remaining time for the print (##:##R) and the remaining time for color change (##:##C). However, the remaining time for color change isn't showing in firmware 3.14.0.

**To Reproduce**
  Add color change at specific layer.

**Expected behavior**
  The MK3S should show the remaining time on the LCD screen and keep toggling between the remaining time for print and the remaining time for color change.

3d-gussner, 2024-08-05:
@majedation Thanks for the bug report. It is not an issue with the FW 3.14.0 firmware as injecting `M73 C1` while printing will show on the LCD the `##:##R` and `00:01C`.

During testing I have found that PrusaSlicer 2.8 generates the M73 gcodes but `C` is always 0 which means that **NO** filament changes are expected in normal mode. At the same time the `D` parameter has some value decreasing over time, but this is only shown in Silent mode.

```
M73 P0 R27
M73 C0
M73 Q0 S27
M73 D3
....
M73 P1 R26
....
M73 D2
....
```

I will report this issue to the PrusaSlicer team.

3d-gussner, 2024-08-05:
@majedation Please consider to close this issue as it is not firmware related.

closed 2024-08-05 by majedation

3d-gussner, 2024-08-05:
@majedation FYI found an internal ticket and :crossed_fingers: it will be solved in next PrusaSlicer version

3d-gussner, 2024-11-21:
@majedation FYI it is back working in [Prusa-Slicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1) 
[image]

3d-gussner, 2024-11-21:
@samdurnxn No need to open it here as it is a Prusa Slicer issue. I will send the new info to PrusaSlicer team.

Can you share the 3mf file where it still fails? Does that work in and older version of PS?

3d-gussner, 2024-11-21:
I am not sure if PrusaSlicer ever calculated the Time to change with MMU prints. Do you have an example where it worked?