prusa3d/Prusa-Firmware #4740: [FEATURE REQUEST] Filament change end of layer

animoose, 2024-07-31:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** -MK4
**MMU Upgrade** - None

**Is your feature request related to a problem? Please describe.**
Selecting "Filament Change" from the Tune menu during a print causes the filament change to happen immediately. I would like to be able to request a filament change at the end of the current layer, similar to the programmed filament changes that can be created using PrusaSlicer. This is useful for the case when you start a long print, then realize you forgot to insert a color change when doing the slicing.

**Describe the solution you'd like**
Under "Change Filament" in the tune menu, there should be two options: change now, and change at end of current layer. An additional possibility might be "change at height" plus the specific height at which the change is wanted.

gudnimg, 2024-08-03:
Hi 👋 I see you mention MK4 as printer type. This repository is only for the 8-bit printers (MK2.5 and MK3/S/+ models).

The MK4, MINI, and XL printers are located here https://github.com/prusa3d/Prusa-Firmware-Buddy/issues

Even if this is for the 8-bit firmware, it is not feasible to implement this as the printer has no concept of when a layer starts or ends. Can't comment on the 32-bit firmware though.

animoose, 2024-08-03:
My mistake - I will re-file it.

closed 2024-08-03 by animoose