prusa3d/Prusa-Firmware #4743: [ENHANCEMENT]<Toolchange Path Prusa XL>

RymdmanSpiff, 2024-08-04:
**Printer type** - Prusa XL

**Is your enhancement related to a problem? Please describe.**
   When a toolchange is done on the Prusa XL the printer moves the tools across the bed. I have the problem that I use TPU as a second material and after purging the TPU there is strings left behind on the printbed. I need high temp on the TPU to get a good bond and then it become very stringy unfortunately.

**Describe the solution you'd like**
  Could the printer move the tools around the outer perimeters of the bed when it pick up and leave new tollheads? It will take a little longer time but there will be no srings on the first layer.

Perhaps it could be a choise in the slicer? "Perimeter toolpath" or "Quickest toolpath" 

//RymdmanSpiff

3d-gussner, 2024-08-05:
@RymdmanSpiff This is the firmware repository for the 8-bit printers MK2.5/S and MK3/S/+ please close this issue here and open it in the 32-bit MINI, MK3.5, MK3.9, MK4 and XL repo https://github.com/prusa3d/Prusa-Firmware-Buddy

closed 2024-08-05 by RymdmanSpiff

3d-gussner, 2024-08-05:
@RymdmanSpiff Thanks for closing it :hugs: