prusa3d/Prusa-Firmware #4764: Power panic communication from power supply?

loughkb, 2024-08-30:
### Question

I live off grid and on solar power most of the year. I will be replacing the printer's power supply with a 12 to 24 volt DC converter for the sake of efficiency. 30 to 40% reduction in power use versus using an inverter and the AC power supply. 
I need to know how the power panic communication from the power supply works so I can duplicate it. This is on a MK4S.

3d-gussner, 2024-09-01:
@loughkb This is the Prusa Firmware repository for the MK2.5, MK2.5S, MK3 and MK3S.

I am not an electronics designer but here some links that might help you with your project.

The MK3 with the Silver PSU had a special PCB which detected the AC power outage and triggered the PowerPanic connector.
See https://github.com/prusa3d/MK3_Power_Panic
The new Delta PSU which is also used on the MK4S has this/similar circuity already on the PSU PCB see here https://github.com/prusa3d/MK3_Power_Panic/blob/master/rev._Delta_PMZ-24V240WDAL/Delta_PMZ-24V240WDAL.pdf

The power panic circuity detects the AC side going down and signals to the printer this event. In the capacitors of the AC PSU is still some power left to trigger the power panic and execute some instructions on the printer controller board. You can imagine that this time slot is very narrow.

So you gonna need to figure out how you can keep the 24V up for that needed time the printer needs to execute the power panic.

3d-gussner, 2024-09-02:
@loughkb Glad you find this information helpful :heart_eyes:
Wish you all the best with your project and maybe it is an idea to share with others later (not here maybe in the forums or printables).

Please close this issue, to keep the ticket count low and manageable.

P.s.: Even when closed please post here your solution, I personally am very interested in your solution and approach.
Side note: There is a 24V UPS project from BTT which may help you aswell. I never used it as I think it have been better to use 10 times 2.7 capacitors instead of only 9 capacitors. 9 x 2.7V = 24.3V which can be exceeded by some PSUs and you should "never" run CAP at 100% if you want them to last long. The other issue with this design is that it may not work with the PP of the Einsy or Buddy controller board.

3d-gussner, 2024-09-02:
@loughkb Please triple check the signal, voltages etc. from the Delta PSU to the `ac_fault` connector on the buddy. And then ensure that your circuit behaves similar and doesn't blow up something on the buddy board.

closed 2024-09-02 by loughkb

loughkb, 2024-09-02:
No worries on the ac_fault connection.  There is no signal voltage coming from the Delta.  That transistor is acting as a switch.  There is a signal voltage coming from the buddy board that it grounds to zero when there is an loss of AC fault.  Probably a  logic line going to an GPI/O pin on a micro-controller