prusa3d/Prusa-Firmware #4770: [QUESTION] MMU2S BOOTLOADER

shz-fpv, 2024-09-09:
### Question

Encountered address is out of range when burning MMU2S BOOTLOADER using atmelflip,The compilation system I used is debian12.1.0, and I directly compiled the entire project using the make command

3d-gussner, 2024-09-12:
@shz-fpv Thanks for the question.

You tried to compile the Prusa-Firmware and got `address is out of range when burning MMU2S BOOTLOADER` ?

This is the repository for the Prusa 8-bit printers not for the MMU which can be found here https://github.com/prusa3d/Prusa-Firmware-MMU

You can't flash the printer firmware/bootloader, which is for a ATMega2560, on a MMU board, which has Atmel32u4. See https://github.com/prusa3d/MM-control-2.0?tab=readme-ov-file and https://github.com/prusa3d/Arduino_Boards

shz-fpv, 2024-09-13:
Thank you very much for your reply. Previously, I did misunderstand the project name. Now, I have successfully compiled the MM-corol-01 project using Linux according to the link you provided, generating MM-corol-01 hex. However, when using atmelFlip to upload firmware, it reported the lnvalid HEX file syntax to me. I am not quite sure where the problem lies