prusa3d/Prusa-Firmware #4775: [BUG] [BUG] "Z-leveling enforced" during mesh bed leveling after upgrading to 3.14.0

sinjindog, 2024-09-18:
### Printer model

MK3S/+

### MMU model

No-MMU

### Firmware versions

3.14.0

### Upgrades and modifications

None

### Printing from...

SD Card

### Describe the bug

I just upgraded to 3.14.0 on a stock MK3S+.  I an unable to do mesh bed level with the new firmware.  I am also unable to complete a z calibration without errors as well.  Rolling back to 3.12 or 3.13 resolves the issue.  7 x 7 is enabled.   It seems like the tolerances are too tight for the levelling to complete.  I have had zero issues with printing over the last 2.5 years with this machine.  The odd time during the 7 x 7 levelling at the start of a print the pinda may stop and take a second reading on the odd spot but I have great first layers and no other issues.  I saw the closed issue where a z calibration fixed the problem but I am unable to complete that without errors as well.  Anyone else having this problem?  Thanks

### How to reproduce

Unable to mesh bed level or do a z calibration with 3.14.0  Other firmwares resolve the issue.

3d-gussner, 2024-09-19:
@sinjindog Thanks for the report, but this is a duplicate of https://github.com/prusa3d/Prusa-Firmware/issues/4762 and is addressed in next FW 3.14.1.

sinjindog, 2024-09-19:
Thanks I was aware it was a dupe but didn't see mention that it would be
corrected in 3.14.1 only that the solution was to do a z calibration which
failed on my printer.  Sorry if I offended.  I'll keep quiet next time.

3d-gussner, 2024-09-20:
> Sorry if I offended.  I'll keep quiet next time.
@sinjindog Nobody is offended and you should NOT be quiet next time. We are just asking to search for similar issues and read what the comments. As you can see in that closed ticket the user was able to solve the issue.

Please try until we release FW 3.14.1
1. Change to MBL 3x3
2. Run Z calibration NOT XYZ calicartion
3. Change back to 7x7

Let me know if that solves your issue.

So please continue to engage and report your issues or ask questions. Even this is an duplicate we appreciate the interaction.

sinjindog, 2024-09-20:
>    2. Run Z calibration NOT XYZ calibration

Thanks for the reply I should have mentioned that I did try the 3 x 3 after
deleting the previous presets just to try and get a completed z calibration
after the normal 7 x 7 failed but it did not work either.  Only z
calibration was ever done, not xyz calibration.  Changing back to 7 x 7
would not do anything as the 3 x 3 calibration failed as well so there were
no values saved.  I'm on 3.13.x that's working fine, I think I'll wait and
test 3.14.1 when it comes out before spending any more time on it.

3d-gussner, 2024-09-20:
@sinjindog Please send me an email.

closed 2024-11-27 by github-actions[bot]