prusa3d/Prusa-Firmware #4776: [FEATURE REQUEST] Re-Enable load filament when filament is already loaded

Tycho-X, 2024-09-20:
### Printer model

- [X] MK3S/+
- [X] MK3
- [ ] MK2.5S
- [ ] MK2.5

### Describe the feature

In the older firmware versions i could use "load filament" even when filament already had been loaded. the 3.14 versions seems to have this option disabled. Please re-enable this option. I use it for various reasons like checking if the filament is dry enough and "flushing" filament which had been exposed to humidity between dry box and print head during night.

### Expected functionality

Re-enable "load filament" option when filament is already loaded. Like in pre 3.14 firmware versions.

Tycho-X, 2024-09-20:
I changed this to bug as i cant load any filament anymore. Doesnt matter if filament is loaded or not.  

See also https://forum.prusa3d.com/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/firmware-3-14-problem-on-i3mks/

3d-gussner, 2024-09-20:
@Tycho-X Thanks for the request.
We implemented this as some users where "complaining" that there are load and unload menus and they clicked per accident the wrong one.

As you can see we can't make everyone happy.

As a workaround you can heatup the nozzle and use the Settings -> Move axis -> Extruder and flush your nozzle.

skandinaff, 2024-09-29:
In my case, I use filament sensor for jam or runout detection, but I keep autoload function off. and in that case the insert filament used to be available in the main menu. Now, it is not, and I need to go to settings, disable filament sensor, then insert filament and reenable it back again. This is kinda frustrating to be honest. I would imagine that making a single menu "Load/Unload" filament with next page specifically asking what to do could keep all people happy.

3d-gussner, 2024-09-30:
@skandinaff I will assign this ticket to myself and discuss your request. Can't promise anything.

drewsed, 2024-10-18:
Any update on this?

I'm tired of putting the filament on, aiming, clicking “Load Filament”, and pushing the filament in quickly. I have never confused “Load” and “Unload”. 
Why does the menu entry change when the filament sensor is active?