prusa3d/Prusa-Firmware #4777: [BUG] Filament runout not working after upgrading to 3.14.0 (I had to disable/reenable Fil. Runout)

adamdport, 2024-09-22:
### Printer model

MK3S/+

### MMU model

MMU2S

### Firmware versions

3.14.0

### Printing from...

SD Card

### Describe the bug

I had a print fail because I ran out of filament but it didn't alert me. I know the filament sensor is working because it's been autoloading filament when I push it in the top of the extruder. I went to `Settings -> Fil. Sensor` and in that menu, `Fil. Sensor` was on, but `F. Runout` and `F. Autoload` were set to `N/A`. I went to `Fil. Sensor` and turned it off and back on again, and THEN `F. Runout` and `F. Autoload` were set to `On`. **I no longer have a problem**, but this was a pretty frustrating experience with a lot of wasted filament, so if it's an easy fix please do so.

Other notes: I DID reinstall 3.13.3 recently as part of #4762 and then re-upgraded to 3.14.0 but I don't know if it's related. Again, to be clear, I no longer have an issue, but I shouldn't have had to disable/reenable `Fil. Runout` to make my printer work. Thanks!

3d-gussner, 2024-09-24:
@adamdport Thanks for the report. I have been looking at the [changes made from FW3.13.3 and FW3.14.0](https://github.com/prusa3d/Prusa-Firmware/compare/v3.13.3...v3.14.0) and can't find any obvious changes that could explain this issue.

I understand that you
1. Downgraded to FW3.13.3 to fix the Z calibration issue
2. Upgraded to FW3.14.0
3. Autoloaded the filament
4. Started a print
5. had a filament runout which didn't trigger
6. Checked after the failed print the Settings -> Fil. sensor
   - `Fil. sensor` was `ON`, but `F. runout` and `F. autoload` and  showed `N/A`
7.  After turning off and on the Fil. sensor
   - `Fil. sensor`, `F. runout` and `F. autoload` showed `ON`

The `N/A` is only shown when the Fil. sensor state is in `error`.
The `error` state is only triggered when the Voltage check fails on the IR sensor 0.4

Please verify
- which FSensor version you have. Look for the [rev version](https://www.prusa3d.com/product/ir-filament-sensor/)
  - LCD -> Support 
- Check wiring [IR Filament Sensor Troubleshooting (MK2.5S, MK3S)](https://help.prusa3d.com/article/ir-filament-sensor-troubleshooting-mk2-5s-mk3s_112226)
   - Check cable for potential break.
      - Go to LCD -> Support -> Sensor info and move your cable loom
      - Go to LCD -> Support -> Voltages
         - insert, remove filament multiple times
         - move cable loom while watching the IR voltage

An other option is to "rerun" the FSensor detection LCD -> Settings -> HW Setup -> Fsensor Detection

While printing you can check the Tune menu as see if Fil. Sensor settings are okay.

Hope that helps.

3d-gussner, 2024-09-30:
@adamdport Thanks for the feedback. Sometimes it is hard to reproduce these kind of issues. Yes please re-open this when it comes back.

>or maybe it's just an intermittent hardware issue looming

Please check if the IR sensor connectors and cables.