prusa3d/Prusa-Firmware #4778: [BUG] Mesh bed levelling grid offset

Gixy31, 2024-09-22:
### Printer model

MK3

### MMU model

No-MMU

### Firmware versions

3.14

### Upgrades and modifications

None

### Printing from...

SD Card

### Describe the bug

Since several upgrades, the mesh bed levelling grid is offset by sevral cm to the left, i.e no more centered on the bed. This has been already signaled in the forum, but still not answered. It is clearly a firmware bug, probably an incorrect offset parameter for this type of printer. Please correct this as the bed levelling is thus inconsistent.

### How to reproduce

nothing to do, just perform mesh bed levelling on a MK3 printer.

### Expected behavior

Grid centerd on the bed, which is no more the case.

Gixy31, 2024-10-31:
Video sent to support team, without reaction, no answer here neither, what a useless community...

Gixy31, 2024-12-31:
Just an update to avoid closing...

3d-gussner, 2024-12-31:
@Gixy31 I assigned this issue to myself so the stale bot will not close it.
As this is not a trivial issue, we gonna need some time to look at it.

3d-gussner, 2024-12-31:
I just printed a 250x210x0.2 two perimeter square (no bottom or top layers) without a brim or skirt.
On my MK3S the
- left line is about 3mm from the sheet edge
- right line is about 1mm from the sheet edge

My XYZ cal. details
Y distance from min
Left: 11.83mm
Right: 11.80mm 
Skew 0.07°
[0;0] point offset
X 0.44mm
Y 1.33mm

3d-gussner, 2024-12-31:
Printed a tiny 0.8 x 0.8 x 6 "tower" centered on the bed using the MK404 sim. To monitor the position I used `M155 S1 C5` to autoreport the temperature and positions. The tiny moves of this tower where very very close to the center by 0.xx mm. The differences are related to the XYZ cal values that the MK404 SIM has.

Gixy31, 2024-12-31:
Hi,
The problem isn't related to normal printing, but only to the mesh bed leveling function, which is offset to the left of more than 2cm! Printing is OK, but the right part of the sheet is never measured, even with the 7x7 grid. In the 7x7 configuration, I even had situation where there were only 6 columns...

3d-gussner, 2025-01-03:
@Gixy31 Thanks for clarifying the issue. We will have a look into this.

3d-gussner, 2025-01-03:
I went back to v3.12.3, v3.9.3 and even back to v3.8.0 and the Mesh bed leveling always ends at X 205 position. With the PINDA x offset off 23 the PINDA is at 228 X position.

1/9 X at 1 --> PINDA X 24
2/9 X at 103 --> PINDA 126
3/9 X at 205 --> PINDA 228

I don't understand when did this change?

Also the PINDA at 2/9 is at X 126 which is nearly middle of the 250 on X. Very good compromise between print area 250 and X axis ~255 length (Belt test).
250 / 2 = 125
~255 / 2 = 127.5
Avg      =  126.25

>Since several upgrades, the mesh bed levelling grid is offset by sevral cm to the left, i.e no more centered on the bed.

Gixy31, 2025-01-04:
Exactly the same here. In fact, the grid is centered compared to the PINDA position, as in the past, I don't remember up to which firmware version, it was centered compared to the nozzle position, say something like 23 at 1/9 and 227 at 3/9, so PINDA at 46 at 1/9 and 250 at 3/9.
Now I understand that it is probably what is wanted and that the previous behaviour was not correct, as what is important is the PINDA position, not the nozzle, right?

3d-gussner, 2025-01-07:
@Gixy31 The PINDA measures at positions 24 , 126,  228 on X and  6, 108, 210 on Y. These values represent best the needed Mesh for the nozzle and the bed.

Also the magnets are an issue and the positions of the probing points are limited see also https://github.com/PavelSindler/Prusa-Firmware/pull/1

I used the MK404 SIM and `G80` (3x3 and 7x7) and `M420` to get the mesh for this example, as these values are reproducible, while real PINDA values will very a bit.

The measured points are different between 3x3 and 7x7 BUT the difference to the center of the bed are identical with both methods (front-left , left , rear-left, rear, rear-right, right , front right, front).

[image]

mdjurfeldt, 2025-01-27:
@3d-gussner @Gixy31 Just a piece of information: I'm using the 7x7 setting on an MK3S+. When the feature to only do mesh bed levelling on the area under the model was first introduced, it worked as expected. With the latest firmware upgrade (3.14.1) I get the (model-limited) bed levelling grid shifted 2 cm towards the origin, both in X and Y direction. The model is then printed at the correct location. (I use PrusaSlicer 2.9.0 if that matters.)

JoneKone, 2025-03-24:
Now I'm not sure if my "issue" is the same on my machine MK3 There was this weird behavior of changing the positions of the mesh bed leveling measurement points. The setting/positions changed all by it self, first it measured 9x9 then it measured other variations like on the very center and 3x3 I ques. Up untill now I didn't even know there was a setting for changing it's behavior. I didn't change any settings and other locations, only by resetting the machine it resumed the original behavior.

It happened all by it self on 3.13.0-6873 and after I updated that to the very latest 3.14.1 today, after calibration it started to measure the big grid.

Like I said dunno if this is related or just user error.
Also sorry for my bad English =) and Grammare

JoneKone, 2025-03-24:
I see, and yeah I think it was 7x7. I will monitor the situation. I was looking at it and commented about the strange behavior and my dad also noticed the different behavior. Unfortunately I didn't keep logs on my procedure or what I was doing... So My comment is totally based on memory of yesterday. And I had other things in my mind.. Maybe I will log my future interactions and try to reproduce the issue.

closed 2025-06-02 by github-actions[bot]