prusa3d/Prusa-Firmware #4787: [BUG]

mi3dke, 2024-10-04:
### Printer model

MK3S/+

### MMU model

No-MMU

### Firmware versions

3.14

### Printing from...

SD Card

### Describe the bug

I consider it unneccessary to heat up the bed for unloading the filament.

### How to reproduce

Choose unload in the printer menu. 
Nozzle and bed are heated up.,

### Expected behavior

For unloading the filament, only the nozzle should be heated.
The bed does not need to be heated up.

Save time and energy.

Panayiotis-git, 2024-10-05:
Hi mi3dke, since [3.13](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.0), there is a setting to control if the bed is also heated during filament load/unload commands.
Settings > HeatBedOnLoad [Yes / No]

mi3dke, 2024-10-05:
Ah, thanks. Will try immediately after the current print finished. 

I should start reading the release notes ;)

mi3dke, 2024-10-05:
closing: already implemented

closed 2024-10-05 by mi3dke

3d-gussner, 2024-10-07:
@mi3dke > I should start reading the release notes ;)

You could :wink: 

@Panayiotis-git Thanks for replying, always appreciated when community members help each other and interact in the github issues.