prusa3d/Prusa-Firmware #4788: [BUG]

sethtroisi, 2024-10-12:
### Printer model

MK3

### MMU model

No-MMU

### Firmware versions

3.14.0-8066

### Upgrades and modifications

None

### Printing from...

SD Card

### Describe the bug

"FW crash detected" each time SD card is removed. 

### How to reproduce

1. Have SD card inserted (inserting SD card works fine)
2. Remove SD card

### Files

[crash.webm](https://github.com/user-attachments/assets/5d95a5b8-310b-43ea-8927-724275038604)

3d-gussner, 2024-10-14:
@sethtroisi I see that you selected in the issue printer model the `MK3` without a `S`, but in your video your printer has `MK3S` firmware installed. Which printer do you have?

I can't reproduce your issue. I had to use the MK404 Simulator as I don't have a real `MK3` at the moment.
See 
https://github.com/user-attachments/assets/66deb3f3-7615-4597-8356-f0c625395367

Did it work with FW 3.14.0 and suddenly it started to crash?

Did you have the issue before FW 3.14.0?

sethtroisi, 2024-10-18:
It's an MK3s.
I recently upgraded to FW 3.14.0 and immediately started having crashes which I had never had before.

3d-gussner, 2024-10-18:
> It's an MK3S. I recently upgraded to FW 3.14.0 and immediately started having crashes which I had never had before.
> 
> I flashed back to `prusa3d_fw_MK3S_MK3S+_3_13_3_7094` and the crashes stop happening. I flashed `MK3S_MK3S+_FW_3.14.0_MULTILANG` again and crashes resume.

I tried to reproduce this issue also on real printer and it works.

Can you please:
1. Use a different fresh formatted FAT32 SDCard
2. Connect via USB and use a serial monitor like Putty or https://www.serialterminal.com/ to capture the serial output.
3. Please provide also the CRASH DUMP see https://help.prusa3d.com/article/crash-dump_364959
4. Contact LiveChat support and ask them to forward your logfiles to me.

sethtroisi, 2024-10-29:
With a different freshly formatted FAT32 2GB SDCard with one partition and only a single file (benchy.gcode)

With 3.13.3, no crash
```
# Insert SD Card
echo:SD card released
LCD status changed
Sending 0xFF
echo:SD card ok
LCD status changed

# Removing SD Card
echo:SD card released
LCD status changed
echo:SD init fail
LCD status changed
```

On 3.14.0
```
start                                                                                                                                                            
echo: 3.14.0-8066_b97131bd2                                                                                                                                 
MMU is Off                                                                                                                                                       
SpoolJoin is Off                                                                                                                                                 
echo: Last Updated: 2024-05-22 05:21:06 | Author: prusa3d                                                                                                        
echo: Free Memory: 2511  PlannerBufferBytes: 1760                                                                                                                
echo:Stored settings retrieved                                                                                                                                   
adc_init                                                                                                                                                         
Hotend fan type: NOCTUA                                                                                                                                          
CrashDetect DISABLED                                                                                                                                             
echo:SD init fail                                                                                                                                                
//action:dump_available                                                                                                                                          
echo:SD init fail

# Insert SD Card
echo:SD card released
LCD status changed
Sending 0xFF
echo:SD card ok
LCD status changed

# Remove SD Card
echo:SD card released
LCD status changed
Sending 0xFF
Error:volume.init failed
LCD status changed
start
echo: 3.14.0-8066_b97131bd2
```

It looks like the printer is believes that an SD card is present when it's not.
This matches the physical experience with the printer where I occasionally have to use a little extra pressure or re-seat the SD card to get it to notice it's inserted.

This still looks like a regression in 3.14 to crash when 3.13.3 didn't.

I sent a crashdump to live support and asked them to send it to you.

closed 2025-01-06 by github-actions[bot]