prusa3d/Prusa-Firmware #4789: [BUG] Logitec C920 wont connect with prusa connect

Kaylaheart, 2024-10-12:
### Printer model

MK3S/+

### MMU model

MMU3

### Firmware versions

6.1.3+7898

### Printing from...

PrusaConnect

### Describe the bug

so. yesterday my MK3s+ with MMU2 upgraded to MK4s with MMU3, webcam Logitech HD Pro Webcam C920. stopped working i reached out to Prusa help but they just pointed to the fact that they havent updated the list for the MK4s compatible webcams and left it as that. anyone else who has this issue please let me know i will try to fix this issue and report back what i find. this is listed as MK3S/+ model but its for MK4S models. next time im buying a BAMBU LABS 3D printer. not this overpriced garbage.

### How to reproduce

connect camera with Logitech HD Pro Webcam C920 on prusa connect

closed 2024-11-01 by Prusa-Support

Kaylaheart, 2024-11-04:
then why did prusa help say it was?

Kaylaheart, 2024-11-04:
and its not a RPI- Camera

Prusa-Support, 2024-12-28:
> then why did prusa help say it was?

Please check the relevant steps in the instructions linked above.
You will find a short description of the compatibility conditions as well as a link to the list of _known_ compatible cameras.
> and its not a RPI- Camera

If a camera is listed then it should be safe to call it "_RPi-compatible_", hence _compatible for MK2.5/3 with Prusa Link via Raspberry boards_, whereas `Raspberry board ≠ Buddy board`, and RPi boards are not required for MK4/XL/MINI.

Everything looks correctly stated in the Help section to me but, if anything looks incorrect or unclear, please add a comment _directly in the Help section_ under the relevant guide/step.

This issue should be considered invalid as per https://github.com/prusa3d/Prusa-Firmware/issues/4789#issuecomment-2452053758.

Michele Moramarco
Prusa Research

Kaylaheart, 2024-12-28:
> This issue should be considered invalid as per #4789 (comment)

You realise the camera is hooked up to my computer not the printer itself
right? The purse connect stopped working after I refreshed the prusa
connect page. I don't understand how after upgrading my printer from mk3 to
mk4 and using the camera successfully with the mk4 using prusa connect it
suddenly without explanation just stopped connecting.