prusa3d/Prusa-Firmware #4791: [ENHANCEMENT] Move the Z-axis up after aborting print for better extruder access

drygoatair, 2024-10-15:
### Printer model

- [X] MK3S/+
- [ ] MK3
- [ ] MK2.5S
- [ ] MK2.5

### Describe the enhancement

The only reason I abort a print job is because the print has failed and I imagine this is true in most cases for most people. A failed print usually means that the print made contact with the hot-end, sticks to the hot-end and is subsequently dragged around by it. The result is a hot-end smeared with sticky hot filament, which now need to be cleaned.

When aborting a print job, the extruder moves out of the way but remains at pretty much the same Z-level. So, when you want to preform the logical next step, which is check if the hot-end needs cleaning, you have to go trough several menu's to move the Z-axis up before the extruder is accessible. To me, this just doesn't make sense. 

People abort print jobs when they fail. When a print fails, the extruder should be accessible and clearly visible so the user can clean it (or, at the very leas, check whether cleaning is necessary). 

### Expected functionality

After a print job is aborted, the extruder moves to 75%-100% on the Z-axis to allow easy inspection and maintenance of the hot-end.

drygoatair, 2024-10-15:
Heh... do I feel stupid 😅 This will do, thanks!

drygoatair, 2024-10-15:
Not necessary.

closed 2024-10-15 by drygoatair

3d-gussner, 2024-10-16:
@drygoatair 
>Heh... do I feel stupid

Don't feel that way. Glad this is an acceptable workaround for you.