prusa3d/Prusa-Firmware #4793: Extruder Linearity Correction for MK3.5(s)

0uincy, 2024-10-18:
### Printer model

- [ ] MK3S/+
- [ ] MK3
- [ ] MK2.5S
- [ ] MK2.5

### Describe the feature

I used linearity correction for the extruder motor on my MK3s+ before upgrading to MK3.5s. I went to look for it in the MK3.5 settings page and noticed that it isn't present on the current firmware.

I saw a significant improvement to extrusion consistency when I used it on my MK3s+ and would like to see it added to a future version of the MK3.5 firmware.

3d-gussner, 2024-10-18:
@0uincy As you updated the MK3S to MK3.5 please close this issue here and open it in the 32-bit [Prusa-Firmware-Buddy repository](https://github.com/prusa3d/Prusa-Firmware-Buddy) .

0uincy, 2024-10-18:
> @0uincy As you updated the MK3S to MK3.5 please close this issue here and open it in the 32-bit [Prusa-Firmware-Buddy repository](https://github.com/prusa3d/Prusa-Firmware-Buddy) .

Oops. Thank you

closed 2024-10-18 by 0uincy