prusa3d/Prusa-Firmware #4796: [ENHANCEMENT] Log error messages

ssackett, 2024-10-19:
### Printer model

- [X] MK3S/+
- [ ] MK3
- [ ] MK2.5S
- [ ] MK2.5

### Describe the enhancement

The printer should log errors so that they are easily retrieved by the user.  This pertains to the MK4.  Some errors appear on the printers screen but once the screen is cleared by resetting or power cycle there is no record of the error.  If the operator didn't take proper notice of the error and calls tech support the error can't be properly described because the evidence of it has disappeared.  I have found that tech support doesn't have a list of the errors to compare with what the user remembers and vague descriptions are not helpful.

### Expected functionality

Provide a user accessible log of errors that duplicates the information displayed by the printer when the error occurred.

3d-gussner, 2024-11-21:
@ssackett Thanks for the request. I doubt that we can squeeze this into the firmware as the resources are very limited.
Using PrusaLink and PrusaConnect shows some history.

3d-gussner, 2024-12-18:
Closing due to no interaction.

closed 2024-12-18 by 3d-gussner