prusa3d/Prusa-Firmware #4802: [ENHANCEMENT] Implement a Load Test All function

natn, 2024-10-27:
### Printer model

- [ ] MK3S/+
- [ ] MK3
- [ ] MK2.5S
- [ ] MK2.5

### Describe the enhancement

As a printer operator, I would like to be able to initiate a loading test for all filaments in the MMU3 without having to select each test individually.

### Expected functionality

A menu item under Loading Test exists called Test All. 
When the Test All option is selected, a filament test will be initiated sequentially for each MMU filament.
If a loading error occurs, the process stops and an error message indicating what filament failed to properly load will be displayed.

3d-gussner, 2024-10-28:
@natn This is the Prusa firmware repository for the MK2.5/S and MK3/S/+ printers. Please close your issue here and open it in the https://github.com/prusa3d/Prusa-Firmware-Buddy which is for the MINI, MK3.5/S, MK3.9/S, MK4/S and XL.

I assume that you have a MK3.5/S, MK3.9/S or MK4/S with an MMU3 connected as the menu on the MK3S/+ Settings->Loading Test has the option "All".

3d-gussner, 2024-12-18:
Closing due to no interaction.

closed 2024-12-18 by 3d-gussner