prusa3d/Prusa-Firmware #4803: [BUG] No way to unload filament anymore if ended ups to screen "Changing filament"

jupe, 2024-10-27:
### Printer model

MK3S/+

### MMU model

No-MMU

### Firmware versions

6.1.3+7898

### Upgrades and modifications

None

### Printing from...

PrusaLink

### Describe the bug

After several hours of print, my filament roll off. I think it asked that "is filament was unload successfully" and I accidentally clicked "yes" and it went to "Changing filament" screen. ->

[image]

I realise that filament wasn't unloaded and there was no way to unload it anymore from this screen. After trying filament unloading manually and trying navigate menu, I ended up to click reset button that rebooted my prusa and breaks printing. Luckily my print was ~ready so I didn't lose anything. 

### How to reproduce

printing until filament roll off and go to above screen without successful filament unload.

### Expected behavior

Would be better UX if there could be way to go back to unload filament even accidentally ended up above screen without successful filament unload.

3d-gussner, 2024-10-28:
@jupe This is the Prusa firmware repository for the MK2.5/S and MK3/S/+ printers. Please close your issue here and open it in the https://github.com/prusa3d/Prusa-Firmware-Buddy which is for the MINI, MK3.5/S, MK3.9/S, MK4/S and XL.

jupe, 2024-10-28:
there was actually similar bug already: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3756.

closed 2024-10-28 by jupe