prusa3d/Prusa-Firmware #4806: [BUG] Stop Print doesn't work for prints started with Prusa Link

scriptengine, 2024-11-03:
### Printer model

MK3S/+

### MMU model

No-MMU

### Firmware versions

3.14.0-8066

### Upgrades and modifications

rPI Zero added as per Prusa documentation, rPI has PI camera

### Printing from...

PrusaLink

### Describe the bug

I generally submit all my prints from Prusa Slicer on my PC. Since the 3.14 firmware if I use the machines LCD console to stop a print nothing happens. The only way I can stop a print is using Prusa Link (I've not tried using Prusa Connect). However if I start a print from the SD card, then I can use the console to stop a print.

### How to reproduce

Start a print from the slicer
(Once the print has started) On the printers LCD control unit attempt to stop the print

### Expected behavior

When stop print is selected the print stops!

3d-gussner, 2024-11-04:
@scriptengine Thanks, but I can't reproduce your issue. I have FW3.14.0 + PrusaLink 0.8.1
1. In PrusaLink Storage I start a print
2. Let it heat up and when printing
3. On LCD screen the stop and it stops the print on the printer and in PrusaLink

I have to admit that I have the MK3S connected via USB to my PC running PrusaLink and it is also connected to PrusaConnect.

So can't tell if the printer behaves differently without PrusaConnect.

scriptengine, 2024-11-04:
Thanks for attempting to repo. It sounds like you are doing the same as me and getting different behaviour.  

I wonder if its just me and how I trouble shoot this!

3d-gussner, 2024-11-04:
@scriptengine Please try to slice a small print and test it while watching the print. As said above we have some fixes in FW 3.14.1.

scriptengine, 2024-11-04:
>>Please try to slice a small print and test it while watching the print.
I put a small (3 minute) test print on SD card from the Slicer and printed it succesfully.
I then printed again from SD card and during the print used STOP on the LCD console to stop the print. The command worked first time immedately.

I then sent the same object to the printer from PrusaLink
During the print both the STOP and PAUSE commands on the LCD console would not respond, where as the TUNE command would change the speed%

Did that answer the question?

>>As said above we have some fixes in FW 3.14.1.
Is 3.14.1 stable and available for download? Do you want me to test it?

scriptengine, 2025-01-06:
I've upgraded to a MK4S so this issue no longer affects me and I'm closing it.

closed 2025-01-06 by scriptengine