prusa3d/Prusa-Firmware #4807: [BUG]

arcady-genkin, 2024-11-04:
### Printer model

MK3S/+

### MMU model

No-MMU

### Firmware versions

3.14.0

### Printing from...

OctoPrint

### Describe the bug

Using a Prusa Slicer 2.8.1 inserted a print pause at 3.6mm to insert a clear sheet into the print.
Verified that the gcode had M601 in it:

`;AFTER_LAYER_CHANGE
;3.6
;PAUSE_PRINT
M117 Insert
M601
G1 Z4
G1 X192.594 Y15.898 F10800
G1 Z3.6 F720
G1 E.8 F1500
;TYPE:Support material
;WIDTH:0.55
G1 F3000
G1 X192.349 Y15.927 E.0104
G1 X191.968 Y15.792 E.01704
`

After pausing, the printer turned off the heating element on the hotend, so the hot end cooled down.
After selecting "Resume print" from the menu, the printer did not heat up the hot end, and the target temperature for the hot end remained as 0*C. But the printer started "printing" without heating the hot end.

The extruder was moving around, but I did not notice that the extrusion was being attempted (I have an extruder motor rotation indicator, and it was not moving).

### How to reproduce

Insert a pause into model.
Print, wait for the pause.
Resume printing.

### Expected behavior

The hot end should heat up to the previous temperature, and printing should resume with filament being extruded.

### Files

[gcode_and_model.zip](https://github.com/user-attachments/files/17614128/gcode_and_model.zip)

3d-gussner, 2024-11-04:
@arcady-genkin Thanks for the report. Can it be that the printer was waiting for a long time before you resumed it?
We addressed few issues in FW 3.14.1 to restore the previous temperature.

arcady-genkin, 2024-11-04:
> @arcady-genkin Can it be that the printer was waiting for a long time before you resumed it?

It was a matter of minutes. About 10 minutes the first time it happened. Less than that the second time.

3d-gussner, 2024-12-18:
Closing due to no response and interaction.

closed 2024-12-18 by 3d-gussner