prusa3d/Prusa-Firmware #4808: [BUG] FW 3.14.0 on i3 Mk3 - Filament sensor settings reset after every start of printer

cordilon, 2024-11-04:
### Printer model

MK3

### MMU model

No-MMU

### Firmware versions

3.14.0 for Bondtech Extruder

### Upgrades and modifications

official Bondtech Extruder upgrade, Sunon 4010 fan for hotend, Misumi linear rods

### Printing from...

SD Card

### Describe the bug

since upgrading from firmware 3.13.0 to 3.14.0, the setting for filament sensor on/off gets reset to on with every start of the printer. I'm not using the filament sensor, so this is really annoying and I have to turn it off every time, so the menu options to manually load/unload filament become available.

### How to reproduce

- install firmware 3.14.0 on i3 Mk3 printer (Bondtech version could be the culprit, but I doubt it)
- start printer
- set filament sensor to "off"
- turn the printer off and on again
- check if the filament sensor is still turned "off"

### Expected behavior

settings for filament sensor should persist after turning printer off and on again.

### Files

https://github.com/user-attachments/assets/e6d5680d-c489-4a1f-a6fb-00a9b36b7ef1

3d-gussner, 2024-11-05:
@cordilon I just tried to reproduce your issue on the MK404 SIM using the Prusa_MK3 with Prusa and Bondtech firmware.
I can't reproduce your issue in both versions the FSensor stays OFF after booting.

Please try to factory reset the printer.

As I see that you have multiple Sheets `SmoPLA`defined I would suggest to backup and restore the sheets settings. See [M850](https://reprap.org/wiki/G-code#M850:_Sheet_parameters)
Connect to printer with a serial terminal
Send
```
M850 S0 ; Get Sheet 0 values
M850 S1 ; Get Sheet 1 values
M850 S2 ; Get Sheet 2 values
M850 S3 ; Get Sheet 3 values
M850 S4 ; Get Sheet 4 values
M850 S5 ; Get Sheet 5 values
M850 S6 ; Get Sheet 6 values
M850 S7 ; Get Sheet 7 values
```
Copy/note the reported values for each sheet to later set them again.

In case you want also to Backup/Restore your Filament Statistics you can use the `D3` code.
The `D3` gcode is CASE sensitive and with some serial terminal or host software they may corrupt/overwrite the EEPROM.
For example Pronterface and OctoPrint auto uppercase terminal inputs.
More information about the [EEPROM](https://prusa3d.github.io/Prusa-Firmware-Doc/Prusa-Firmware/group__eeprom__table.html)

To backup Statistics
```
D3 Ax0ff1 C4 ; Read EEPROM_FILAMENTUSED
D3 Ax0fed C4 ; Read EEPROM_TOTALTIME
```
Copy note the results and you can later restore these values.
AGAIN `D3` is case sensitive and the Address need to start with `Ax` and NOT `AX` as the uppercase `X` is used to write values.

Example:
Sending `D3 Ax0ff1 C4` with 0.42 m Filament used the output is
```
D3 - Read/Write EEPROM
000ff1  2a 00 00 00
```

To restore the 0.42m used filament after a factory reset you need to send `D3 Ax0ff1 X2a000000`
The values behind the writing `X` must be hex and always a factor of 2 numbers/characters. When the length is uneven the firmware will overwrite the EEPROM and only recovery option is a factory reset.

cordilon, 2024-11-05:
Hi @3d-gussner and thanks for the suggestions. A **full factory reset however did NOT fix the issue**.
Since I'm not using the filament sensor, it wasn't even installed on my printer, but to pass the self check, I temporarily installed it. After that, the settings for filament sensor on/off were saved correctly and stayed the same after a restart of the printer. Only **when I unplugged the filament sensor, it stopped working again**!
As I said, it was working with the sensor unplugged in the preview firmware version, so there must have been some internal changes. Can you reproduce that with the simulation software? I'd do it myself, but the usage seems wildly complicated.

3d-gussner, 2024-11-05:
@cordilon Thanks for the update. I can't find anything in the changes from [3.13.3 to 3.14.0](https://github.com/prusa3d/Prusa-Firmware/compare/v3.13.3...v3.14.0) which could explain that the PAT9125 FSensor is activated again.
Few questions:
You have MK3 or a MK3S with the Bondtech extruder?
Which Bontech extruder do you have?
Which Bontech firmware did you install?

Differences between the MK3 vs MK3S are:
- [PAT9125 FSensor](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/variants/MK3.h#L150) vs [IR FSensor](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/variants/MK3S.h#L152)
- Extruder body height [2](https://github.com/prusa3d/Prusa-Firmware/blob/v3.14.0/Firmware/Marlin_main.cpp#L2241-L2243) vs [9](https://github.com/prusa3d/Prusa-Firmware/blob/v3.14.0/Firmware/Marlin_main.cpp#L2244-L2246)
  - This is difference isn't there in the [Bontech firmware](https://github.com/BondtechAB/Bondtech-Prusa-Firmware/blob/Bondtech-v3.14.0/Firmware/Marlin_main.cpp#L2241-L2245) it is the same for the MK3 and MK3S only changes with the LGX but also there for both.
- [AltFan](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/variants/MK3S.h#L354-L355) is only available on MK3S
- MMU configuration [MK3+MMU](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/variants/MK3.h#L668) vs [MK3S+MMU](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/variants/MK3S.h#L672)

Which filament sensor did you connect temporary? [PAT9125](https://help.prusa3d.com/guide/5-e-axis-assembly-textile-sleeve_71288#77792) or [IR Sensor](https://help.prusa3d.com/guide/5-e-axis-assembly_28536#29560)

As the MK3 PAT9125 sensor is an I2C device and the IR senors an analog device I could imagine that the firmware doesn't like that the I2C module isn't connected. A quick look doesn't show any changes to the I2C, so it is just an idea.

As you don't have a filament sensor connected I would suggest that you flash the Bondtech MK3S firmware to eliminate potential issues with the I2C communication.

Also a "floating" cable without any connection can cause some issues on serial communication ports.

As an example for floating cables causing issues is when you have the MMU cable connected to the EINSY board but NOT to the MMU, this can cause unexpected issues on the printer.

>Can you reproduce that with the simulation software?

This would need some changes in the MK404 SIM, which I will not try to do.

cordilon, 2024-11-05:
Hi @3d-gussner and thanks again for the quick reply.
I have a Prusa i3 Mk3 from 2018.
I have the official "Prusa I3 MK2.5-MK3 Extruder upgrade" installed [ https://www.bondtech.se/product/prusa-i3-mk2-5-mk3-extruder-upgrade/ ]. Seems to be based on BMG extruder, but since it is specifically designed to fit the old carriage and E3D hotend, it doesn't have any other name.
I installed the 3.14.0 firmware from the Bondtech website [ https://github.com/BondtechAB/Bondtech_Prusa_i3/raw/master/Firmware/3.14.0/FW3140-Build8370-Bondtech-MK3-EINSy10a-E3Dv6full.zip ] and I always pay extra attention not to accidentally use the Mk3S version.
Here are photos of the filament sensor I have, seems to be the PAT9125.
[image]
[image]

> I would suggest that you flash the Bondtech MK3S firmware to eliminate potential issues with the I2C communication.

Wouldn't it cause other problems to flash Mk3S firmware onto a Mk3?

> Also a "floating" cable without any connection can cause some issues on serial communication ports.

For the last couple of years I had the whole sensor plus cable uninstalled and it was working fine.

3d-gussner, 2024-11-05:
> Wouldn't it cause other problems to flash Mk3S firmware onto a Mk3?

Not if you DON'T connect the PAT9125 MK3 Sensor.
A disconnected MK3S IR Sensor shouldn't cause any issues.
I just unplugged my IR sensor and the Support -> Sensor shows `Fil. sensor 0` and Support -> Voltages `IR : 5.0V` 

> For the last couple of years I had the whole sensor plus cable uninstalled and it was working fine.

I can't find any changes that may have changed the behavior. Sorry. I don't know what else it can be. The printer "should" have the PAT9125 sensor connected to avoid these kind of issues. There have been improvements regarding the MMU which depends on the filament sensor and there might be the issue you are having.

As the MK3 is so old I doubt that I can spend more time to investigate this issue with a disconnected PAT9125. Sorry.

I just run the MK404 SIM with the IR sensor "not" responding and ran the Wizard after a factory reset.
- Wizard told me that I needed to remove filament but continued
- During the first layer cal I was asked to load the filament and could finish the calibration and wizard.

So I can even more suggest that you flash the MK3S firmware and disable the FSensor.

I am surprised that the Bondtech MK3 had NO PAT9125 solution at the time. How can you have a 3D printer without a filament runout sensor?

Alternative would be to get https://support.bondtech.se/Guide/03.%20Sensor%20Add-On%20for%20MK3s/30.html and get https://www.prusa3d.com/product/ir-filament-sensor/ + https://www.prusa3d.com/product/ir-filament-sensor-einsy-cable/

cordilon, 2024-11-05:
Ok, I'll keep flashing Mk3S firmware in mind as an option. For now I'll probably leave the old filament sensor dangling out of the electronics box. At the moment I also don't see the point paying 20€ and printing new parts for a feature that I don't really use. I always calculate the needed amount of filament before a print and make sure it's available. The filament sensor is like parking sensors on a car to me: It's a neat safety feature, but optional if you know what you're doing.
I'd just downgrade to 3.13.0 again, but the latest version actually has some features I want to use.

3d-gussner, 2024-11-06:
@cordilon Valid point not to buy the IR sensor etc. But I still think you are better of with the MK3S firmware and nothing connected than having the PAT9125 dangling out of the box. If you still want to keep the PAT9125 connected at least print a small cover to protect the board and prevent it from floating around.

You had FW 3.13.0 before? I always compared FW 3.13.3 to FW3.14.0.

Please consider to close this issue.

cordilon, 2024-11-06:
@3d-gussner the sensor is already wrapped in foam and tucked between other cables, so it should be safe. As mentioned: I'll look into flashing Mk3S firmware and re-doing the calibration when I have the time.

Yes, I upgraded from 3.13.0 to 3.14.0, since there were no significant features in between that were important for me. If you have the time, maybe you can find differences there. I'll wait for your response and then close the issue.

3d-gussner, 2024-11-06:
The changes from 3.13.0 to 3.14.0 are even bigger and get really hard to find issues.
If you have the time and still want to use the MK3 firmware can you try to update from 3.13.0 to 3.13.1 to 3.13.2 and 3.13.3 to see when this issue happens.

cordilon, 2024-11-06:
@3d-gussner I won't have the time to flash all the firmwares myself, but read through the change log and found this for 3.13.2:

> **Enable/Disable FSensor using G-codes**
> This release brings support for enabling and disabling filament sensor (FSensor) using G-codes M405/M406 submitted by @Commod0re

I could see that this somehow changed the behavior of the filament sensor in general. Is there a way on your part to take a look at that specifically?

Commod0re, 2024-11-06:
Review my PR:

https://github.com/prusa3d/Prusa-Firmware/pull/1447

I only edited one file having to do with executing gcode, so it's doubtful this caused any significant change in operational behavior

gudnimg, 2025-01-12:
From the video in the original post, I can see that the filament sensor is **enabled** and that the lines below are marked `N/A.` This means the filament sensor is in **error state**. While the filament sensor is in error state, the firmware considers the sensor to be enabled.

We sometimes only use `fsensor.isEnabled()` and don't check if its in error condition. For example in the new filament sensor warning in 3.14.1.

[image]

Note the line` MENU_ITEM_TOGGLE_P(_T(MSG_FSENSOR), fsensor.isEnabled() ? _T(MSG_ON) : _T(MSG_OFF), lcd_fsensor_enabled_set);` doesn't use the EEPROM value to detemine if ON or OFF is shown. This means if the sensor is in error state, the menu will show ON regardless of what is in EEPROM. 

[image]

cordilon, 2025-01-12:
@gudnimg Thanks for your input, this seems to be an important clue. I still don't understand why it would be in error state if the sensor is disconnected, especially when it wasn't happening in earlier firmware versions. Do you have suggestions?

gudnimg, 2025-01-12:
@cordilon I suspect the issue was introduced here in 9f81156. FYI @3d-gussner. this causes the bug to appear and block showing Load Filament and Unload filament menus. This commit only appears in 3.14.0 and newer.

The underlying issue has been present in 3.13.0 though, it just didn't impact your work from what I can tell. So to fix the underlying issue we need adjust `fsensor.isEnabled()` to check if the filament sensor state machine is in error state.

I'm using this description as reference, it really help find this 
> since upgrading from firmware 3.13.0 to 3.14.0, the setting for filament sensor on/off gets reset to "on" with every start of the printer. I'm not using the filament sensor, so this is really [...]

[image]

I'll see if I can create pull request for this tonight.

FrHePo, 2025-11-02:
I have the same issue wit my modded MK3. I don't have a filament sensor in my extruder (https://www.printables.com/model/249267-extruder-for-prusa-mk3-with-e3d-revo-micro-and-orb) and I usually load my filament manually. After starting a print I get the message that no filament is loaded. Setting the filament-sensor to OFF works but after a new machine-start the filament-sensor is switched on again. Firmwareversion is 3.14.1

FrHePo, 2025-11-03:
Unknown M code: M412 S0 (Flag to enable or disable Filament Runout Detection).
So I'll have to downdate to 3.14.0

L928, 2026-01-30:
I can't fully test it since I don't have an IR sensor.
But I also extend the feature to use a mech. switch instead of IR.
I am still on it.

FrHePo, 2026-01-30:
Yes, it's still a nuissance although not a grave one. The filament-setting should at least be saved for the next printer-start. I'm not using a filament-sensor (meaning there is no filament sensor in my printer) so why is the system setting this switch to 'on' anyway?

L928, 2026-02-08:
Seems #4834 solved it, but i don't see if it had been pulled into the mian branch yet.

FrHePo, 2026-10-19:
It is correct that Prusa is ignoring this issue?