prusa3d/Prusa-Firmware #4812: [BUG] Printing from USB

steve1515, 2024-11-15:
### Printer model

MK3S/+

### MMU model

No-MMU

### Firmware versions

3.14.0

### Upgrades and modifications

None

### Printing from...

OctoPrint

### Describe the bug

When printing from USB, the LCD screen used to show "USB" in previous firmware versions, but now shows "HO". What does "HO" mean? This seems like a bug or mistake. I wanted to point it out in the event that it was an inadvertent mistake.

Here's the relevant change: https://github.com/prusa3d/Prusa-Firmware/commit/20a2216623c2d54eb8c58d9a27a6cec6e0000624#diff-5fbfdca43c192573e31d8e78d255c510fb6e213b118cf38173277ec2a2163880R362

### How to reproduce

Print from USB.

### Expected behavior

The LCD screen to say USB or COM. Or, an explanation as to what "HO" means in this context.

sarusani, 2024-11-16:
HO stands for Host (USB Host). The layout changed and now only two characters are used for identifying the source of the print.
HO for USB. SD for SDCard, PL for PrusaLink and if you use the [OctoPrint plugin](https://plugins.octoprint.org/plugins/ActivatePrusaHostTimer/) OP for OctoPrint.

3d-gussner, 2024-12-18:
@steve1515 Closing this issue as @sarusani explained the meaning of `HO`.

By changing from `USB`  to two letter also frees up a character on the LCD screen for future improvements.

Here a short summary of known host short names:
- `HO` = Host. Default when Gcode commands are received via serial and not set by M79 S"<<hosttype:2>>"
- `PL` = PrusaLink. PrusaLink sends `M79 S"PL"` when it starts up and no PrusaConnect is configured.
- `PC` = PrusaConnect. PrusaLink sends `M79 S"PC"` when it is connected to PrusaConnect.
- `OP` = OctoPrint. The OctoPrint Plugin mentioned by @sarusani sends `M79 S"OP"` to keep the host timer active.

closed 2024-12-18 by 3d-gussner