prusa3d/Prusa-Firmware #4818: [BUG] Busy Status on printer

Elcar903, 2024-12-10:
### Printer model

MK3S/+

### MMU model

MMU3

### Firmware versions

3.14.1-8237

### Upgrades and modifications

none

### Printing from...

PrusaConnect

### Describe the bug

Printer shows "Busy" status on Prusa connect when its not busy and you cant start a print either from the app or from the manual controls on the printer.

Using PrusaLink  0.8.1 and MMU3 (3.0.3)

I have tried to re-flash the firmware to 3.14.1 (3.0.3) on both the printer and MMU3 unit  and this doesn't fix it.
I downgraded the firmware to 3.14.0 (3.0.2) and it fixes the issue.

Sometimes resting the printer with the "X" button will fix it or resetting just the MMU3 will fix it but not always.

### How to reproduce

Just turning on the printer and it goes into the Busy state.

### Expected behavior

turn on the printer and being able to use it without going into busy state when its not doing anything.

gudnimg, 2024-12-16:
Did the printer visually indicate in any way _why_ its busy? Was there any message on the last line on the screen? Or was the printer showing perhaps a full screen message? 

Usually the printer is only in busy state on boot-up if the printer is waiting for user feedback.

Elcar903, 2024-12-17:
No message on the screen, everything looks normal.

I let it sit over 30 min after boot up with no changes to the status.
It wont even let me print from the SD card when it shows "Busy" on Prusa connect.

Re-flashing the firmware to Firmware 3.14.1 (3.0.3) to see if the first time it didn't do soothing correctly 
on both the MK3s and MMU3 as well as doing a factory reset on both devices after 
the re-flash did not fix the issue either.

Some times after several resets/power cycles it will show correctly and works just fine,

I downgraded the firmware back to Firmware 3.14.0 (3.0.2)  and everything works perfectly with no issues like before.

I have spent at least 6 hours messing with everything I can think of,
Checking hardware, wires, connectors , etc.  and followed all the trouble shooting guides and nothing has fixed it for me.

3d-gussner, 2024-12-18:
@Elcar903 Thanks for the update and your time troubleshooting it
We have been troubleshooting this issue as well and I have found last night what is causing it.

Short summary:
PrusaLink 0.8.1 tries to get the MMU firmware version and expects the response in a very strict format. While all previous MMU version matched that format the MMU3.0.3+896 frankly doesn't.

We are working on a solution.

Would you be willing to try a "hot-fix"?

Elcar903, 2024-12-18:
@3d-gussner I wouldn't mind trying it.

Paulg5678, 2024-12-19:
I'm seeing this issue as well, was working fine after firmware update until i did a power cycle, now permanently stuck in BUSY mode. 

Is the current workaround just to flash back to the old firmware until the 0.8.2 prusaLink is released?

3d-gussner, 2024-12-20:
@Elcar903 @Paulg5678 Please send me an email.

saskot, 2024-12-20:
Hi i am having the same issue MK3S mmu2s. happend with update to 3.14.1-8237 also did update on rasp. System Version

Python

3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0]

3d-gussner, 2024-12-21:
@saskot See https://github.com/prusa3d/Prusa-Firmware/issues/4818#issuecomment-2551953529 or send me an email.

mgtmadness141879, 2025-01-20:
I am getting the same. MK3s+, MMU3 , newest update of 3.14.1-8237 
I was also getting the error: job id cannot be obtained.
It currently shows "Busy" on Prusa Connect and Prusa Link, app and browser.

3d-gussner, 2025-01-21:
@mgtmadness141879 I update the PrusaLink PR and added a manual [workaround](https://github.com/prusa3d/Prusa-Link/pull/1009#issuecomment-2603678526).
If you still have issues please send me an email.

mgtmadness141879, 2025-01-21:
> @mgtmadness141879 I update the PrusaLink PR and added a manual [workaround](https://github.com/prusa3d/Prusa-Link/pull/1009#issuecomment-2603678526).
> If you still have issues please send me an email.

Email will be sent shortly

Naugrimohtar, 2025-02-01:
I too am experiencing this issue, can you also email me instructions?

Naugrimohtar, 2025-02-01:
I believe I have applied the changes. When might the 0.8.2 with this change roll out?

3d-gussner, 2025-02-10:
Please read https://github.com/prusa3d/Prusa-Link/pull/1009#issuecomment-2603678526

> When might the 0.8.2 with this change roll out?

I don't know but keep you posted as soon I know more.

Elcar903, 2025-02-10:
Going into settings on the mk3 after the prusa link on the mk3 screen says ready and turning mmu off then waiting for about 10 seoncds and then turning the mmu setting back on has been working for me as a fix but still annoying to do this everyday.

MichaelN1967, 2025-02-22:
i have same problems with my MK3S+ - MMU3. I have connected Printer working with Pi Zero W to Connect / Prusa Link. It works perfect before i install latest FW for Printer and MMU3
Don´t have SSH installed on my Image for Raspi. What can i do to fix the problem ?

S-Sizelove, 2025-02-25:
After updating to the latest fimrware on my MK3s+ and MMU3 with PrusaLink installed on a RPI zero2w I also have the same problem.... think you could get to fixing the naming convention on the MMU3 or releasing an update to PrusaLink 0.8.1 so that my printer will work correctly? this BUG was apparently first reported December 10th 2024 and it is now Febuary 25 2025.  seems like a long time to leave this problem open...

3d-gussner, 2025-02-26:
@S-Sizelove Please read https://github.com/prusa3d/Prusa-Link/pull/1009#issuecomment-2603678526 if you still have issue send me an email.

closed 2025-05-05 by github-actions[bot]